YosysHQ/yosys #2915: Exception std::out_of_range' on mapping BRAM cells for libtrellis (Lattice ECP5)

janschiefer, 2021-08-04:
## Steps to reproduce the issue

Affected version: yosys GIT current

Error:
` 
(...)
    Rule for bram type $__ECP5_DP16KD (variant 1) rejected: requirement 'attribute syn_romstyle="ebr" ...' not met.
  Checking rule #6 for bram type $__ECP5_DP16KD (variant 2):
    Bram geometry: abits=11 dbits=9 wports=0 rports=0
    Estimated number of duplicates for more read ports: dups=1
    Metrics for $__ECP5_DP16KD: awaste=1589 dwaste=4 bwaste=16137 waste=16137 efficiency=75
    Rule for bram type $__ECP5_DP16KD (variant 2) rejected: requirement 'attribute syn_romstyle="ebr" ...' not met.
  Checking rule #6 for bram type $__ECP5_DP16KD (variant 3):
    Bram geometry: abits=12 dbits=4 wports=0 rports=0
    Estimated number of duplicates for more read ports: dups=1
    Metrics for $__ECP5_DP16KD: awaste=3637 dwaste=0 bwaste=14548 waste=14548 efficiency=70
    Rule for bram type $__ECP5_DP16KD (variant 3) rejected: requirement 'attribute syn_romstyle="ebr" ...' not met.
  Checking rule #6 for bram type $__ECP5_DP16KD (variant 4):
    Bram geometry: abits=13 dbits=2 wports=0 rports=0
    Estimated number of duplicates for more read ports: dups=1
    Metrics for $__ECP5_DP16KD: awaste=7733 dwaste=0 bwaste=15466 waste=15466 efficiency=52
    Rule for bram type $__ECP5_DP16KD (variant 4) rejected: requirement 'attribute syn_romstyle="ebr" ...' not met.
  Checking rule #6 for bram type $__ECP5_DP16KD (variant 5):
    Bram geometry: abits=14 dbits=1 wports=0 rports=0
    Estimated number of duplicates for more read ports: dups=1
    Metrics for $__ECP5_DP16KD: awaste=7733 dwaste=0 bwaste=7733 waste=7733 efficiency=52
    Rule for bram type $__ECP5_DP16KD (variant 5) rejected: requirement 'attribute syn_romstyle="ebr" ...' not met.
  Selecting best of 6 rules:
    Efficiency for rule 4.5: efficiency=52, cells=32, acells=1
    Efficiency for rule 4.4: efficiency=52, cells=32, acells=2
    Efficiency for rule 4.3: efficiency=70, cells=24, acells=3
    Efficiency for rule 4.2: efficiency=75, cells=20, acells=5
    Efficiency for rule 4.1: efficiency=83, cells=18, acells=9
    Efficiency for rule 1.1: efficiency=88, cells=17, acells=17
    Selected rule 1.1 with efficiency 88.
    Mapping to bram type $__ECP5_PDPW16KD (variant 1):
      Read port #0 is in clock domain \VexRiscvLitexSmpCluster_Cc1_Iw32Is4096Iy1_Dw32Ds4096Dy1_ITs4DTs4_Ood_Wm.bufferCC_3.debugCd_external_clk.
        Mapped to bram port B1.1.
terminate called after throwing an instance of 'std::out_of_range'
  what():  vector::_M_range_check: __n (which is 32) >= this->size() (which is 32)
build_ulx3s.sh: Zeile 3: 82994 Abgebrochen             (Speicherabzug geschrieben) yosys -l ulx3s.rpt ulx3s.ys
`

Compile the attached verilog file with the following build script:

`
verilog_defaults -push
verilog_defaults -add -defer
read_verilog /jschiefer_shortfn/litex/pythondata-cpu-vexriscv-smp/pythondata_cpu_vexriscv_smp/verilog/Ram_1w_1rs_Generic.v
read_verilog /jschiefer_shortfn/litex/pythondata-cpu-vexriscv-smp/pythondata_cpu_vexriscv_smp/verilog/VexRiscvLitexSmpCluster_Cc1_Iw32Is4096Iy1_Dw32Ds4096Dy1_ITs4DTs4_Ood_Wm.v
read_verilog /jschiefer_shortfn/litex/linux-on-litex-vexriscv/build/ulx3s/gateware/ulx3s.v
verilog_defaults -pop
attrmap -tocase keep -imap keep="true" keep=1 -imap keep="false" keep=0 -remove keep=0
synth_ecp5   -json ulx3s.json -top ulx3s
`

## Expected behavior

Normal synthesis.

## Actual behavior

Terminated because of C++ exception  'std::out_of_range' in
  what():  vector::_M_range_check: __n (which is 32) >= this->size() (which is 32)

[ulx3s.zip](https://github.com/YosysHQ/yosys/files/6934914/ulx3s.zip)
[verilog_files.zip](https://github.com/YosysHQ/yosys/files/6934915/verilog_files.zip)

mwkmwkmwk, 2021-08-05:
Already fixed in #2909, please update.

closed 2021-08-05 by mwkmwkmwk

janschiefer, 2021-08-05:
Great!