YosysHQ/yosys #2916: active low enable for tristate buffer

pepijndevos, 2021-08-05:
@yrabbit found that Gowin tristate buffers "OEN" inputs are active low, which I confirmed, but `synth_gowin` does not treat them as such.

https://github.com/YosysHQ/yosys/blob/2e421feb0ea526468493ab008f3c72beb12c2bc6/techlibs/gowin/synth_gowin.cc#L282

To reproduce, take the following code and synthesize it with Yosys and the Gowin IDE. You will see the vendor produces an IBUF wired straight to a TBUF, while `synth_gowin` generate a `LUT1` inverter between the two.

```
`default_nettype none
`define LEDS_NR 3
// TBUF
module bidir(
    inout wire io_pin,
    input wire i_output,
    input wire data_to_pin);

    assign io_pin = (i_output == 1) ? data_to_pin : 1'bz;
endmodule

//
module top (
    input wire clk,
    input wire key_n,
    inout wire [`LEDS_NR-1:0] led
);

wire [`LEDS_NR-1:0] w_dir;

bidir bd[`LEDS_NR-1:0](led, w_dir, 1'b1);

assign w_dir[`LEDS_NR-1:1] = {`LEDS_NR-1{1'b0}};
// switch the direction with the key
assign w_dir[0] = !key_n;
endmodule
```

I don't see any support in `iopadmap` for this. But it could be it needs to happen at `$_TBUF_` inference or somewhere else.

What's the correct way forward?

gatecat, 2021-08-05:
You can use the same "temporary cell and techmap" trick as Nexus and Xilinx do:

https://github.com/YosysHQ/yosys/blob/2e421feb0ea526468493ab008f3c72beb12c2bc6/techlibs/nexus/synth_nexus.cc#L336

https://github.com/YosysHQ/yosys/blob/2e421feb0ea526468493ab008f3c72beb12c2bc6/techlibs/nexus/cells_map.v#L56

pepijndevos, 2021-08-05:
My concern with that is, at least in the current flow iopadmap happens quite late so there isn't much opportunity to optimize away the inverter. But I guess that's what the `run("opt -fast");` line is for. Thanks, I'll give it a go.

mwkmwkmwk, 2021-08-05:
This is exactly why it's called so early in the Xilinx flow — move it likewise in the Gowin flow.

pepijndevos, 2021-08-05:
I think there were some issues moving stuff around because the Gowin vendor tools are particularly picky, but I guess with some massaging it should indeed work. Might have to play a bit with splitnets or the bits option to make sure there are no wide nets in the output.