YosysHQ/yosys #2926: Conflicting init values for signal 1'0 (seen in LiteX build for Fomu)

tcal-x, 2021-08-10:
## Steps to reproduce the issue

Origin: LiteX build for Fomu; in litex-boards/litex_boards/targets/, execute `./kosagi_fomu.py --cpu-variant=minimal --build`

From the attached zip archive: unzip, enter gateware/, and execute `bash build_kosagi_fomu_pvt.sh`.

[fomuerror.zip](https://github.com/YosysHQ/yosys/files/6963591/fomuerror.zip)

Yosys ident: `Yosys 0.9+4251 (git sha1 681a1c07, clang 11.0.1-2 -fPIC -Os)`

Output:
```Adding SRST signal on $auto$simplemap.cc:442:simplemap_dffe$12287 ($_DFFE_PP_) from module kosagi_fomu_pvt (D = $procmux$5360.Y_B [4], Q = \main_spiflash_phy_shift_cnt [4], rval = 1'0).
Adding SRST signal on $auto$simplemap.cc:442:simplemap_dffe$12286 ($_DFFE_PP_) from module kosagi_fomu_pvt (D = $procmux$5360.Y_B [3], Q = \main_spiflash_phy_shift_cnt [3], rval = 1'0).
Adding SRST signal on $auto$simplemap.cc:442:simplemap_dffe$12285 ($_DFFE_PP_) from module kosagi_fomu_pvt (D = $procmux$5360.Y_B [2], Q = \main_spiflash_phy_shift_cnt [2], rval = 1'0).
Setting constant 0-bit at position 0 on $auto$opt_dff.cc:744:run$30900 ($_SDFFE_PP0P_) from module kosagi_fomu_pvt.

4.39.9. Executing OPT_CLEAN pass (remove unused cells and wires).
Finding unused cells or wires in module \kosagi_fomu_pvt..
ERROR: Conflicting init values for signal 1'0 (\main_basesoc_cdcusb_cdcusbphy_inhandler_dtbs [4] = 1'0 != 1'x).
```

## Expected behavior

Yosys would complete without error.   This older yosys does not encounter this error, using the same design sources: `Yosys 0.9+4081 (git sha1 45968ad7, clang 11.0.1-2 -fPIC -Os)`.   Of course it could be that it truly is a source error, and the current yosys is correctly catching it.

## Actual behavior

The error shown above, `Conflicting init values for signal 1'0`.   From the information provided, I am unable to match it back to the source Verilog.

tcal-x, 2021-08-10:
Sorry I missed #2920 -- this seems to be a duplicate of that.

mwkmwkmwk, 2021-08-12:
Yep, duplicate.  Also, should be fixed in current yosys.

closed 2021-08-12 by mwkmwkmwk