YosysHQ/yosys #2965: ecp5: LiteX dual clock memory may fail to map to DP16KD

gregdavill, 2021-08-15:
## Steps to reproduce the issue

Consider this example where we read/write with different clocks. This is the pattern used by LiteX. 
Since 1eea06bcc0750de02a460f3e949df2f68f800382 it does not correctly map to DP16KD's if the output does not end up registered. 

Run the following verilog module through synth_ecp5
```
yosys -p "read_verilog test.v" -p "synth_ecp5"
```

```verilog
module memtest(clk_a, clk_b, wr, wr_addr, wr_value, rd_addr, rd_value);

input clk_a, clk_b, wr;
input [8:0] wr_addr, rd_addr;
input [15:0] wr_value;
output wire [15:0] rd_value;

reg [15:0] mem0 [0:512];
reg [8:0] rd_addr_r;

always @(posedge clk_a) begin
    if (wr)
        mem0[wr_addr] <= wr_value;
end

always @(posedge clk_b)
    rd_addr_r <= rd_addr;

assign rd_value = mem0[rd_addr_r];

endmodule
```

## Expected behavior

Before 1eea06bcc0750de02a460f3e949df2f68f800382 I got this result

```
   Number of cells:                  1
     DP16KD                          1
```

## Actual behavior

After The mentioned commit above, and in the current git master I get the following
```
   Number of cells:               3344
     L6MUX21                       561
     LUT4                         1781
     PFUMX                         865
     TRELLIS_DPR16X4               128
     TRELLIS_FF                      9
```

In the output from yosys it appears to not pick up the output clock domain `\clk_b`.
```
2.26. Executing MEMORY_BRAM pass (mapping $mem cells to block memories).
Processing memtest.mem0:
  Properties: ports=2 bits=8208 rports=1 wports=1 dbits=16 abits=10 words=513
  Checking rule #1 for bram type $__ECP5_PDPW16KD (variant 1):
    Bram geometry: abits=9 dbits=36 wports=0 rports=0
    Estimated number of duplicates for more read ports: dups=1
    Metrics for $__ECP5_PDPW16KD: awaste=511 dwaste=20 bwaste=18416 waste=18416 efficiency=22
    Rule #1 for bram type $__ECP5_PDPW16KD (variant 1) accepted.
    Mapping to bram type $__ECP5_PDPW16KD (variant 1):
      Shuffle bit order to accommodate enable buckets of size 9..
      Results of bit order shuffling: 0 1 2 3 4 5 6 7 8 9 10 11 12 13 14 15 -1 -1
      Write port #0 is in clock domain \clk_a.
        Mapped to bram port A1.
      Read port #0 is in clock domain !~async~.
        Bram port B1.1 has incompatible clock type.
        Failed to map read port #0.
```

mwkmwkmwk, 2021-08-15:
Good, breaking this pattern was exactly the point — the transformation of register on address input into a synchronous read port is unsound when the memory is written in a different clock domain, and it results in completely changing the behavior of the port, 
along with a simulation mismatch.

Use this pattern to make a synchronous read port instead:

```
always @(posedge clk_b)
  rd_value <= mem0[rd_addr];
```

gregdavill, 2021-08-15:
Okay good to know it was a deliberate change, rather than a regression.

You're right about the pattern. 
I think it's supposed to create a write first memory. But that doesn't make sense when you have an independent read clock.

I've been able to update my project to create the pattern you've described.

closed 2021-08-15 by gregdavill