YosysHQ/yosys #2966: readmem[hb] spuriously called with module parameter default value

q3k, 2021-08-15:
## Steps to reproduce the issue

```
$ yosys -V
Yosys 0.9+4052 (git sha1 687f381b69, gcc 10.3.0 -fPIC -Os)
$ cat top.v
`default_nettype none

module Parametrized(input wire clk, output wire led);
    parameter FILENAME = "";

    reg [31:0] rom [0:8191];

    initial begin;
        $readmemh(FILENAME, rom, 0, 8191);
    end

    always @(posedge clk) begin
        led <= rom[0][0];
    end
endmodule

module top(input wire clk, output wire led);
    Parametrized #("/tmp/foo") pm(clk, led);
endmodule
$ yosys -Q -p "synth_ice40 -top top" top.v

-- Parsing `top.v' using frontend `verilog' --

1. Executing Verilog-2005 frontend: top.v
Parsing Verilog input from `top.v' to AST representation.
Generating RTLIL representation for module `\Parametrized'.
top.v:0: ERROR: Can not open file `` for \$readmemh.
```

## Expected behavior

The design should IMO synthesize, with the top.led wire being set to some value dependent on `/tmp/foo`.

I am not sure what the standards say here, but this is something I've discovered when using Bluespec's standard library with Yosys, eg. in [BRAM1Load.v](https://github.com/B-Lang-org/bsc/blob/main/src/Verilog/BRAM1Load.v#L16).

## Actual behavior

Yosys attempts to `readmemh("", ...)` and fails - ie., seems to do a `readmemh` with the default parameter value of FILENAME. This happens even when `top` doesn't have a `Parametrized` instance, so it seems to be done at a very early pass when `Parametrized` is being read and parsed as a module, before it gets instantiated into `top` as `pm`.

whitequark, 2021-08-15:
Try:

    yosys -Q  -p "read_verilog -defer top.v" -p "synth_ice40 -top top"

q3k, 2021-08-15:
Yeah, that works.

Did I miss some obvious spot where this is documented? Wonder how we can prevent others from stumbling upon this…

thoughtpolice, 2021-08-19:
Just wanted to say, thanks for posting this ticket at least! This has always been a nagging issue of mine when using **[yosys-bluespec](https://github.com/thoughtpolice/yosys-bluespec)** to load Bluespec code; packages like `BRAM::*` simply wouldn't work, and my only workaround until now was this ungodly hack using `connect_rpc` and some regex that I never put upstream. `-defer` works much better, it seems. I've migrated yosys-bluespec to use it in the mean time to see if it can work reliably, and I've updated the packages in Nixpkgs to include all this.

I suspect there are still some other issues with this approach though; in order to do recursive loading of Bluespec primitives, I analyze the AST to see what modules it needs to load. Since `-defer` means that the AST isn't elaborated until a later `hierarchy` pass, I can't just look at the AST after calling `read_verilog`, so imagine this:

- `bsc` emits `Foo.v`, which we load
- `Foo` needs a `CoolBluespecPrim` cell, which we confirm by looking at the missing cells
- Load `$BLUESPECDIR/Verilog/CoolBluespecPrim.v` using `-defer`
- `CoolBluespecPrim` now needs `OtherCoolModule`, but this isn't discoverable: `-defer` means the tree is 'empty' and we can't find missing cells until later. If we didn't use `-defer` in step 3, we could see this.

So I think this case will fail. But I think the Bluespec primitive modules don't expose this pattern, so it might be OK(???) Really, I want a combination of both `read_verilog -defer` to make the above example work + a variant of "run `hierarchy -libdir` really early on in order to auto-load modules from a directory." If I could do that, I think it would solve all the major problems at once. The issue is I don't know how to propagate information like "Use this libdir" from a plugin to the global `hierarchy` pass (so that it works with ordinary `synth` calls, etc.) I suspect some changes would be needed for this...

Honestly I'm not super sure how to proceed here, and I'm simply rambling about my needs, but I just figured I'd brain dump this here since it was a nice discovery last night and it's tangentially related to the original. There are a number of things we could do to fix this in the immediate term for Bluespec at least, but in general I'm not sure what all would need to happen to make this all behave more nicely for everyone and whatnot.

q3k, 2021-08-19:
> CoolBluespecPrim now needs OtherCoolModule, but this isn't discoverable: -defer means the tree is 'empty' and we can't find missing cells until later. If we didn't use -defer in step 3, we could see this.

FWIW what I end up doing with my bluespec/yosys integration (where I just shell out to yosys after compiling Bluespec down to verilog) is just to load all of the Bluespec stdlib into yosys' read_verilog, save for some elements that are broken:

https://code.hackerspace.pl/q3k/qfc/tree/WORKSPACE?h=main&id=0b67dc9f45970c59c62a8b09704729c202fc246e#n42 (this is 'all *.v files *except* `ConstrainedRandom.v`, `Convert*Z.v`, ...')

Doing -defer allowed me to 'unbreak' `BRAM*.v`, maybe some more are now also fixed, haven't checked.

This emits mosnter scripts like the following:

            read_verilog -defer external/bluespec/lib/Verilog/ASSIGN1.v external/bluespec/lib/Verilog/BRAM1.v external/bluespec/lib/Verilog/BRAM1BE.v external/bluespec/lib/Verilog/BRAM1BELoad.v external/bluespec/lib/Verilog/BRAM1Load.v external/bluespec/lib/Verilog/BRAM2.v external/bluespec/lib/Verilog/BRAM2BE.v external/bluespec/lib/Verilog/BRAM2BELoad.v external/bluespec/lib/Verilog/BRAM2Load.v external/bluespec/lib/Verilog/BypassCrossingWire.v external/bluespec/lib/Verilog/BypassWire.v external/bluespec/lib/Verilog/BypassWire0.v external/bluespec/lib/Verilog/CRegA5.v external/bluespec/lib/Verilog/CRegN5.v external/bluespec/lib/Verilog/CRegUN5.v <snip> bazel-out/k8-fastbuild/bin/lanai/Lanai_CPU/mkLanaiCPU.v bazel-out/k8-fastbuild/bin/lanai/CPU_ALU/mkALU.v bazel-out/k8-fastbuild/bin/lanai/CPU_RegisterFile/mkCPURegisterFile.v bazel-out/k8-fastbuild/bin/boards/ulx3s/Board/mkTop.v
            synth_ecp5 -abc9 -nowidelut -top mkTop
            write_json bazel-out/k8-fastbuild/bin/boards/ulx3s/ulx3s.json

Which naturally makes synethesis slower than it could be, but it works :). Doing discovery and loading on demand would be of course nicer, but if anyone's looking for a solution, let it be known that the brute force approach seems to work.