YosysHQ/yosys #2967: Yosys error due to v2 memory cell

antonblanchard, 2021-08-16:
The [Microwatt](https://github.com/antonblanchard/microwatt/) CI has started failing with an error about a `$mem_v2` cell:

```
12.24.5. Executing OPT_CLEAN pass (remove unused cells and wires).
Finding unused cells or wires in module \toplevel..
Removed 2 unused cells and 23 unused wires.
<suppressed ~3 debug messages>
ERROR: Found error in internal cell \toplevel.\soc0.processor.decode2_0.11684 ($mem_v2) at kernel/rtlil.cc:1018:
  attribute \hdlname "soc0 processor decode2_0 11684"
  cell $mem_v2 \soc0.processor.decode2_0.11684
    parameter \ABITS 6
    parameter \INIT 186'000000000001000110110110001000001000000100000111000000000000000011011001010000000000000111000000000000111111101011000000001011011011001001001000010010010000111010010000000000001001111000
    parameter \MEMID "\\soc0.processor.decode2_0.11684"
    parameter \OFFSET 0
    parameter \RD_ARST_VALUE 3'xxx
    parameter \RD_CE_OVER_SRST 1'1
    parameter \RD_CLK_ENABLE 1'1
    parameter \RD_CLK_POLARITY 1'1
    parameter \RD_COLLISION_X_MASK 1'0
    parameter \RD_INIT_VALUE 3'xxx
    parameter \RD_PORTS 1
    parameter \RD_SRST_VALUE 3'000
    parameter \RD_TRANSPARENCY_MASK 1'0
    parameter \RD_TRANSPARENT 1'0
    parameter \RD_WIDE_CONTINUATION 1'0
    parameter \SIZE 62
    parameter \WIDTH 3
    parameter \WR_CLK_ENABLE 1'0
    parameter \WR_CLK_POLARITY 1'0
    parameter \WR_PORTS 0
    parameter \WR_PRIORITY_MASK 1'0
    parameter \WR_WIDE_CONTINUATION 1'0
    connect \RD_ADDR $flatten\soc0.\processor.\decode2_0.$auto$ghdl.cc:762:import_module$5866
    connect \RD_ARST 1'0
    connect \RD_CLK \clkgen.clkop
    connect \RD_DATA \soc0.processor.decode2_0.r [386:384]
    connect \RD_EN $flatten\soc0.\processor.\decode2_0.$auto$ghdl.cc:762:import_module$5770
    connect \RD_SRST $auto$opt_dff.cc:256:combine_resets$23765
    connect \WR_ADDR { }
    connect \WR_CLK { }
    connect \WR_DATA { }
    connect \WR_EN { }
  end
```

I've been trying to provide a cut down test case, but my attempts so far make the problem go away. It requires the ghdl-yosys-plugin, but you can hit it on Microwatt with:

```
make FPGA_TARGET=ORANGE-CRAB YOSYS=/path/to/yosys.install/bin/yosys GHDLSYNTH=/path/to/ghdl-yosys-plugin/ghdl.so microwatt.json
```

antonblanchard, 2021-08-16:
@mwkmwkmwk Let me know if you need more information. I'll try and come up with a better test case.

mikey, 2021-08-16:
FWIW the CI failure related to this is here https://github.com/antonblanchard/microwatt/runs/3336787725#step:3:10663

mwkmwkmwk, 2021-08-16:
Nah, this bug is as obvious as they come. My bad.

closed 2021-08-16 by mwkmwkmwk

antonblanchard, 2021-08-16:
Thanks @mwkmwkmwk! I confirmed it fixes our issue.

PR #2968 (2021-08-16): kernel/mem: Remove old parameter when upgrading $mem to $mem_v2.
Fixes #2967.
commit 10f8b75d (2021-08-16): kernel/mem: Remove old parameter when upgrading $mem to $mem_v2.

Fixes #2967.