YosysHQ/yosys #2972: Running synth_xilinx with abc9 multiple times causes an error

acomodi, 2021-08-20:
## Steps to reproduce the issue

1. Build yosys - `Yosys 0.9+4276 (git sha1 75a4cdfc8, clang 6.0.0-1ubuntu2 -fPIC -Os)`
1. Extract the [tar.gz](https://github.com/YosysHQ/yosys/files/7019796/abc-fail.tar.gz) archive
1. `cd abc-fail && make`

## Expected behavior

The flow should complete without errors.

## Actual behavior

Running `synth_xilinx -abc9` when there are FDPE (or other FF) cells causes the following error:

```
3.44. Executing ABC9 pass.

3.44.1. Executing ABC9_OPS pass (helper functions for ABC9).

3.44.2. Executing ABC9_OPS pass (helper functions for ABC9).

3.44.3. Executing SCC pass (detecting logic loops).
Found 0 SCCs in module top.
Found 0 SCCs.

3.44.4. Executing ABC9_OPS pass (helper functions for ABC9).
ERROR: Attempted to add new module named '\FDPE_$abc9_byp', but a module by that name already exists
Makefile:2: recipe for target 'all' failed
make: *** [all] Error 1

```

eddiehung, 2021-08-31:
Attached test case appears to be corrupted.

eddiehung, 2021-09-01:
Please give #2993 a try.

acomodi, 2021-09-01:
@eddiehung Thanks, I have tested #2993 and the test passes now.

PR #2993 (2021-09-09): abc9: make re-entrant
Clean up some state at the end of `abc9` so that it can be re-entrant.

Fixes #2972.
closed 2021-09-09 by eddiehung