YosysHQ/yosys #2974: CXXRTL: ERROR: Assert `!is_lhs' failed in backends/cxxrtl/cxxrtl_backend.cc:969

ultraembedded, 2021-08-20:
An assert fires in CXXRTL 'Assert `!is_lhs' failed in backends/cxxrtl/cxxrtl_backend.cc:969' with the attached test case.
This happens with the latest yosys codebase.

## Steps to reproduce the issue
[top_v.txt](https://github.com/YosysHQ/yosys/files/7022523/top_v.txt)

yosys -g -p "read_verilog top_v.txt ; write_cxxrtl top.hpp"

## Expected behaviour
No assert :-)

## Actual behaviour
ERROR: Assert `!is_lhs' failed in backends/cxxrtl/cxxrtl_backend.cc:969.

## Version
Yosys 0.9+4276 (git sha1 75a4cdfc, gcc 10.3.0-1ubuntu1~20.10 -fPIC -Os)

commit 75a4cdfc8afc10fed80e43fb1ba31c7edaf6e361 (HEAD -> master, origin/master, origin/HEAD)
Author: github-actions[bot] <41898282+github-actions[bot]@users.noreply.github.com>
Date:   Wed Aug 18 00:51:20 2021 +0000

    Bump version

JulianKemmerer, 2021-09-21:
I can confirm this issue still exists. Seems to pop up if you add debug wires/ports that might optimize away.

JulianKemmerer, 2021-09-22:
Hello there @whitequark - if you have a quick moment could you explain the nature behind the assertion failures at points like `cxxrtl_backend.cc:969 `?

I ask since, per above, I am hitting this assertion as well. 

`TL/DR` assertions seem fine to ignore, top.cpp looks good.

I did some quick console logging experimenting to see if I could find what point in my design the problem is coming from - doing what I think was 'looking at the mangled string c++ code representation of the problem wire'
[image]

getting output like
```
is_lhs==1 skipping INLINE log_assert(!is_lhs); i_auto_24_ghdl_2e_cc_3a_762_3a_import__module_24_4
is_lhs==1 skipping CONST log_assert(!is_lhs)); i_auto_24_ghdl_2e_cc_3a_762_3a_import__module_24_4
is_lhs==1 skipping INLINE log_assert(!is_lhs); i_auto_24_ghdl_2e_cc_3a_762_3a_import__module_24_3
is_lhs==1 skipping CONST log_assert(!is_lhs)); i_auto_24_ghdl_2e_cc_3a_762_3a_import__module_24_3
```

I expected something to fail later - perhaps regarding those wires `i_auto_24_ghdl_2e_cc_3a_762_3a_import__module_24_4`
But things actually completed just fine it seems. Output top.cpp seems to compile and run fine. I dont see any reference to wires like the above in the output cpp so I dont know where they came from or went? (aside from me coming through the ghdl plugin for yosys flow but I cant gather much else to tell you)

Can I safely ignore these? Thanks so much for your time!

whitequark, 2021-09-22:
No, these are not safe to ignore and they may lead to miscompilations.