YosysHQ/yosys #2975: How to use equiv_add command?

mbiewlanz, 2021-08-20:
## Steps to reproduce the issue

Good evening!

I'm trying to launch this command:
"equiv_add csa0.y_gold csa0.y_gate" to execute logic equivalence check: Behavioural Verilog (Golden circuit) vs. Structural Verilog (Revised circuit) 
csa0.y_gold is hierarchical name of the golden wire. 
csa0.y_gate is hierarchical name of the revised wire.

But Yosys returns the error:
ERROR: This command must be executed in module context!

Are there any examples how to use this command? And is it possible to compare logic of Behavioural Verilog model and Structural Verilog netlist in the actual Yosys version?

Thanks for advance!

bphinz, 2022-06-03:
I'm facing the same issue, were you ever able to figure it out?

leeway64, 2025-08-20:
I think you have to select the module that you want first, either by using `cd` or `select`.