YosysHQ/yosys #2976: ecp5: can't map true dual port RAM

thoughtpolice, 2021-08-20:
## Minimal working example

Consider, first off, that I'm not very good at Verilog. Then, consider this:

```systemverilog
// bram_2r2w.sv
`timescale 1 ns / 1 ps
`default_nettype none

module bram_2r2w
  # ( parameter DATA_WIDTH = 32
    , parameter ADDR_WIDTH = 5
    )
    ( input wire logic clk

    , input  wire logic                  wen_1
    , input  wire logic                  wwr_1
    , input  wire logic [ADDR_WIDTH-1:0] addr_1
    , input  wire logic [DATA_WIDTH-1:0] wdata_1
    , output reg        [DATA_WIDTH-1:0] rdata_1

    , input  wire logic                  wen_2
    , input  wire logic                  wwr_2
    , input  wire logic [ADDR_WIDTH-1:0] addr_2
    , input  wire logic [DATA_WIDTH-1:0] wdata_2
    , output reg        [DATA_WIDTH-1:0] rdata_2
    );

  (* syn_ramstyle = "block_ram" *)
  reg [DATA_WIDTH-1:0] ram[0:(1 << ADDR_WIDTH)-1];

  always_ff @(posedge clk) begin
    if (wen_1) begin
      if (wwr_1) begin
        ram[addr_1] <= wdata_1;
        rdata_1     <= wdata_1;
      end else begin
        rdata_1 <= ram[addr_1];
      end
    end
  end

  always_ff @(posedge clk) begin
    if (wen_2) begin
      if (wwr_2) begin
        ram[addr_2] <= wdata_2;
        rdata_2     <= wdata_2;
      end else begin
        rdata_2 <= ram[addr_2];
      end
    end
  end
endmodule
```

This is an attempt to model a faithful dual port RAM for Bluespec, modeled on the original Bluespec **[BRAM2.v](https://github.com/B-Lang-org/bsc/blob/main/src/Verilog/BRAM2.v)** code. While debugging something I decided to try writing my own clean version, and came up with this to minimize things.

## Observed behavior

However, synthesis with a recent yosys does not map this to a proper RAM cell:

```
$ yosys -V
Yosys 0.9+4276 (git sha1 75a4cdfc8a, gcc 10.3.0 -fPIC -Os)

$ yosys -p 'read_verilog -sv bram_2r2w.sv; synth_ecp5' | tail -20
=== bram_2r2w ===

   Number of wires:                 13
   Number of wire bits:            207
   Number of public wires:          13
   Number of public wire bits:     207
   Number of memories:               0
   Number of memory bits:            0
   Number of processes:              0
   Number of cells:                  3
     $mem_v2                         1
     LUT4                            2

2.51. Executing CHECK pass (checking for obvious problems).
Checking module bram_2r2w...
Found and reported 0 problems.

End of script. Logfile hash: d51daf270b, CPU: user 0.49s system 0.08s, MEM: 317.46 MB peak
Yosys 0.9+4276 (git sha1 75a4cdfc8a, gcc 10.3.0 -fPIC -Os)
Time spent: 63% 20x read_verilog (0 sec), 6% 18x opt_clean (0 sec), ...
```

The new memory pass tells us that the second write port wasn't mapped properly, but doesn't say why. (I note that the memory pass does include some code to report reasons for the write port not matching, but not in this case, and I didn't quite investigate further.)

```
  Checking rule #2 for bram type $__ECP5_PDPW16KD (variant 1):
    Bram geometry: abits=9 dbits=36 wports=0 rports=0
    Estimated number of duplicates for more read ports: dups=1
    Metrics for $__ECP5_PDPW16KD: awaste=480 dwaste=4 bwaste=17408 waste=17408 efficiency=5
    Rule #2 for bram type $__ECP5_PDPW16KD (variant 1) accepted.
    Mapping to bram type $__ECP5_PDPW16KD (variant 1):
      Shuffle bit order to accommodate enable buckets of size 9..
      Results of bit order shuffling: 0 1 2 3 4 5 6 7 8 9 10 11 12 13 14 15 16 17 18 19 20 21 22 23 24 25 26 27 28 29 30 31 -1 -1 -1 -1
      Write port #0 is in clock domain \clk.
        Mapped to bram port A1.
      Write port #1 is in clock domain \clk.
        Failed to map write port #1.
    Mapping to bram type $__ECP5_PDPW16KD failed.
```

## Expected behavior

I would expect a `DP16KD` or something? Or that my code is wrong, which I could also reasonably expect. If I remove port 2 entirely (including the parameters), I get at `PDPW16KD` with some flip flops for the read path, at least...

I've also tried this with two clock domains, and no luck with that either.

## Other notes

As I've been watching the new v2 memory cell redesign work going on, I'm wondering how we should interpret lingering `$memrd` and `$mem_v2` cells laying around once a target synth pass is complete. Various little snippets I've tested have teased these out, including a 2r1w RAM I wrote, but that one works just dandy now. Should report any cases where we find lingering `$mem_v2` or `$memrd` cells laying around? Like, should we always at least get LUTRAMs or DFFs? Or should these be interpreted as a warning that memories can't be inferred properly for the target at all, and I should write it differently entirely?

Other than that it's great to see better memory inference happening, so I'm otherwise pleased!

mwkmwkmwk, 2021-08-20:
See #1959. This will be done soon.

thoughtpolice, 2021-08-21:
Oh, I see, I wasn't sure if things were quite settled yet, I was just spinning up an older piece of code to test the latest revision. I'll close this and watch that ticket in the meantime, instead. Feel free to re-open if you think there's anything unique here.

closed 2021-08-21 by thoughtpolice