YosysHQ/yosys #2980: Yosys tcl command substitution does not return the results of the command

macd, 2021-08-23:
The result of a tcl command such as `set xxx [cmd]` should set the variable xxx to the result of command cmd.  This works for normal tcl commands but not for Yosys commands.  Here is a simple example. Say we have test.tcl with the following:
```
yosys -import
set xxx [expr 1 + 2]
puts "variable xxx: $xxx"

set yyy [help zinit]
puts "variable yyy: $yyy"
```
This results in
```
yosys> tcl test.tcl
[TCL: yosys -import] Command name collision: found pre-existing command `cd' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `eval' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `exec' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `read' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `trace' -> skip.
variable xxx: 3

    zinit [options] [selection]

Add inverters as needed to make all FFs zero-initialized.

    -all
        also add zero initialization to uninitialized FFs

variable yyy: 

yosys> 
```

rockybulwinkle, 2022-06-22:
Did you find how to capture the output of yosys commands? I am struggling with this same issue.

rockybulwinkle, 2022-06-23:
Well after much trial and tribulation, I made an incredibly ugly hack to get this functionality.

Basically, It redirects stdout to a temporary file, reads that file, and returns the value. It also restores stdout before returning.

I'm at a loss why I had to make this hack, it seems like basic functionality that should be builtin to yosys. What's the point of TCL scripting if the script can't react to what yosys returns?

Anway, here's the ugly code. I do depend on tclx for access to the "dup" command (for stdout preservation). There is a way to do it without dup (read here: https://stackoverflow.com/a/8532484/600853), but ugly as it is, if we don't preserve stdout, it will mess up logging.

```
#for some reason yosys doesn't initialize basic functionality like the ability to load packages. Do it manually here.
set tcl_library "/usr/share/tcl8.6"
source /usr/share/tcl8.6/init.tcl

#require tclx for dup command
package require Tclx

#import the yosys commands
#yosys -import

proc capture_stdout {args} {
    #open a tmpfile.
    set temp [exec mktemp]
    set newstdout [open $temp w]

    #Save the old stdout to restore later
    set old_stdout [dup stdout]
    #close old stdout
    close stdout

    #make the new file act as stdout
    dup $newstdout stdout

    #Run the yosys command
    {*}$args

    #close the tempfile (acting as stdout)
    close stdout

    #restore stdout fd
    dup $old_stdout stdout

    #read the temp file
    set response [open $temp r]
    set retval [read $response]
    close $response

    #Cleanup
    file delete $temp

    #return what we read
    return $retval
}

#it works with any command. For example:
puts "before"
set capture [capture_stdout "puts" "hello world"]
puts "after"
puts $capture
```

I am emphatically not a TCL expert, there may be some glaring issues in this code. But it works well enough for me...

nakengelhardt, 2022-06-23:
It's not implemented because the TCL interface is a straight import of the Yosys CLI commands, and the Yosys CLI doesn't have conditionals. We usually make a ScriptPass if we need to decide something on contents of the design.

BTW, could you avoid the need for the package if you use the yosys `tee` command to write the output to the temp file instead?

nakengelhardt, 2022-12-05:
With #3572 you can now skip the file on disk and return the output via the scratchpad (still using `tee`, in the future we plan to make commands return values directly where it makes sense).