YosysHQ/yosys #3004: Yosys Verilog frontend: module parameters

SiliconWizard, 2021-09-12:
It looks like Yosys frontend for Verilog is not fully compatible with the Verilog output of GHDL synth when it comes to module parameters.

It chokes on the following:

> module ODDRX1F
>   #(
>     GSR)
>   (input  SCLK,
>    input  RST,
>    input  D0,
>    input  D1,
>    output Q);
> endmodule

with the following error: 

> Test.v:3: ERROR: In pure Verilog (not SystemVerilog), parameter/localparam with an initializer must use the parameter/localparam keyword

I'm not good enough at Verilog to know what Verilog standard the above is compliant with, or if it's even standard Verilog. But GHDL synth generates this kind of module definitions for VHDL components that have generics.

mwkmwkmwk, 2021-09-12:
You can try using yosys in SystemVerilog mode, via the `-sv` option to the Verilog frontend (or via naming the input file something.sv).

Better yet, I recommend using the yosys GHDL plugin instead of going from GHDL to yosys via Verilog.

SiliconWizard, 2021-09-12:
I tried having Yosys use the SV mode, but then it issues another error on the line that instantiates the above module:

> ODDRX1F sdram_clock #(
>     .GSR("01000101010011100100000101000010010011000100010101000100"))
>     (

with the following error:

> Test.sv:689: ERROR: syntax error, unexpected '#', expecting '(' or '['
> Parsing SystemVerilog input from `Colorlight_i5_Test2.sv' to AST representation.

I'll have to try the yosys GHDL plugin, I assumed there was no real difference. Does the plugin not use Verilog as an intermediate?

mwkmwkmwk, 2021-09-12:
Now that looks like plain old incorrect code (the `#(...)`param assignment is supposed to come before `sdram_clock` instance name).

The GHDL plugin goes directly from GHDL IR to yosys IR, it doesn't involve Verilog.

mwkmwkmwk, 2021-09-12:
Also the value given for the parameter is definitely incorrect — it somehow converted an "ENABLED" string into its binary representation, then emitted *that* as a string, there's no way this could possibly work.

SiliconWizard, 2021-09-12:
Thanks! I'll have to report this to the GHDL team I guess.

Using the yosys GHDL plugin, yosys completes successfully.