YosysHQ/yosys #3016: Can't PNR one of the two Icebreaker hardcoded SPI blocks.

71GA, 2021-09-24:
I read [here](https://www.latticesemi.com/view_document?document_id=52206)  (pages 118, 119, 120) all that fllows....

FPGA UP5K SG48 has 2 SPI blocks. Looking from the top view, first SPI block is located in the "lower left" corner and second SPI block is located at "lower right" corner. 

In order to choose the "lower left" SPI block, we first instantiate it and then add one more line like this:
`defparam SB_SPI_inst.BUS_ADDR74 = "0b0000";`

In order to choose the "lower right" SPI block, we first instantiate it and then add a bit different line:
`defparam SB_SPI_inst.BUS_ADDR74 = "0b0001";`

The problem is that in first case place & route (using `nextpnr`) suceeds while in the second case it fails saying:
```
ERROR: Invalid value for BUS_ADDR74 for cell 'SB_SPI_inst' of type 'SB_SPI'
ERROR: Packing design failed.
```

Here is the minimal working verilog design `top.v` (tweak the value `"0b0000"` to  `"0b0001"` to see it fail at the place & route step:

```
module
	top(
		input pcb_oscilator,
		output pcb_rgb_led0,
		output pcb_rgb_led1,
		output pcb_rgb_led2,
		output pcb_led1,
		output pcb_led2,
		output pcb_led3,
		output pcb_led4,
		output pcb_led5,
		input pcb_spi_sck,
		input pcb_spi_csn,
		input pcb_spi_si,
		output pcb_spi_so
);

	wire sb_clki;
	reg sb_rwi;
	reg sb_stbi;
	reg [7:0] sb_adri;
	reg [7:0] sb_dati;
	reg [7:0] sb_dato;
	wire sb_acko;
	wire so;
	wire si;
	wire scki;
	wire scsni;

	SB_SPI SB_SPI_inst(
		.SBCLKI(sb_clki),      // System bus: clock (i)
		.SBRWI(sb_rwi),        // System bus: read/write (i)
		.SBSTBI(sb_stbi),      // System bus: strobe (i)
		.SBADRI0(sb_adri[0]),  // System bus: address to target control registers (i)
		.SBADRI1(sb_adri[1]),
		.SBADRI2(sb_adri[2]),
		.SBADRI3(sb_adri[3]),
		.SBADRI4(sb_adri[4]),
		.SBADRI5(sb_adri[5]),
		.SBADRI6(sb_adri[6]),
		.SBADRI7(sb_adri[7]),
		.SBDATI0(sb_dati[0]),
		.SBDATI1(sb_dati[1]),  // System bus: data input (i)
		.SBDATI2(sb_dati[2]),
		.SBDATI3(sb_dati[3]),
		.SBDATI4(sb_dati[4]),
		.SBDATI5(sb_dati[5]),
		.SBDATI6(sb_dati[6]),
		.SBDATI7(sb_dati[7]),
		.SBDATO0(sb_dato[0]),  // System bus: data output (o)
		.SBDATO1(sb_dato[1]),
		.SBDATO2(sb_dato[2]),
		.SBDATO3(sb_dato[3]),
		.SBDATO4(sb_dato[4]),
		.SBDATO5(sb_dato[5]),
		.SBDATO6(sb_dato[6]),
		.SBDATO7(sb_dato[7]),
		//.SPIIRQ(),           // SPI: interrupt output (o)
		//.SPIWKUP(),          // SPI: wakeup from standby (o)
		.SBACKO(sb_acko),      // System bus: acknowledge (o)
		//.MI(),               // SPI: master input (i)
		.SO(so),               // SPI: slave out (o)
		//.SOE(),              // SPI: slave out enable - active high (o)
		.SI(si),               // SPI: slave in (i)
		//.MO(),               // SPI: master out (o)
		//.MOE(),              // SPI: master out enable - active high (o)
		.SCKI(scki),           // SPI: slave clock in (i)
		//.SCKO(),             // SPI: slave clock out (o)
		//.SCKOE(),            // SPI: slave clock out enable - active high (o)
		.SCSNI(scsni)          // SPI: slave chip select (i)
		//.MCSNO0(),           // SPI: master chip select (o)
		//.MCSNO1(),
		//.MCSNO2(),
		//.MCSNO3(),
		//.MCSNOE0(),          // SPI: master chip select enable - active high (o)
		//.MCSNOE1(),
		//.MCSNOE2(),
		//.MCSNOE3()
	);
	defparam SB_SPI_inst.BUS_ADDR74 = "0b0000";

	// H:
	assign pcb_oscilator = sb_clki;
	assign pcb_spi_sck = scki;
	assign pcb_spi_csn = scsni;
	assign pcb_spi_si = si;
	assign pcb_spi_so = so;

endmodule
```

This is the `makefile` that you can use:

```
filename = top
pcf_file = io.pcf

build:
	yosys -p "synth_ice40 -json $(filename).json" $(filename).v
	nextpnr-ice40 \
		--up5k \
		--package sg48 \
		--json $(filename).json \
		--pcf $(pcf_file) \
		--asc $(filename).asc
	icepack $(filename).asc $(filename).bin
```

You can also use this physicall constraint file `io.pcf` to test :

```
set_io pcb_oscilator 35
set_io pcb_spi_csn 16
set_io pcb_spi_sck 15
set_io pcb_spi_si 17
set_io pcb_spi_so 14

```

gatecat, 2021-09-28:
I believe this is actually a typo in the Lattice docs and the correct address MSBs for the second SPI block is `0b0010`.

71GA, 2021-09-28:
I tested and with the address you suggested, PNR worked!

closed 2021-10-01 by gatecat