YosysHQ/yosys #3074: equiv_struct fail to prove equivalence

jaromharris, 2021-11-10:
[and2.zip](https://github.com/YosysHQ/yosys/files/7509211/and2.zip)
 In this zip file I have a yosys_commands.ys file with the commands I am using to prove equivalence between 2 structurally equivalent verilog netlists.

The netlists are exactly the same, but when running these commands in Yosys I get the following error:
ERROR: Found 4 unproven $equiv cells in 'equiv_status -assert'.

I would expect that it would come back as equivalent.

I suspect I am not setting up the commands correctly, and I would appreciate anyone who can help me with this.

clairexen, 2021-11-22:
The following script proves structural equivalence.

Changes:
- Read Xilinx library as library, so `-flatten` doesn't import the simulation model into the design.
- Run `prep` for both input netlists
- Add `clean` (`;;`) where appropiate

```
read_verilog -lib +/xilinx/cells_sim.v
read_verilog and2_impl.v
prep -flatten
hierarchy -auto-top
rename -top gold
splitnets -ports;;
design -stash gold

read_verilog -lib +/xilinx/cells_sim.v
read_verilog and2_reversed.v
prep -flatten
hierarchy -auto-top
rename -top gate
splitnets -ports;;
design -stash gate

read_verilog -lib +/xilinx/cells_sim.v
design -copy-from gold -as gold gold
design -copy-from gate -as gate gate
equiv_make gold gate equiv
hierarchy -top equiv;;
equiv_struct;;
equiv_status -assert
```

jaromharris, 2021-12-13:
@clairexen thank you for your help!

Follow up question: Is equiv_struct still appropriate to use when comparing equivalence between a 6 input LUT and a 2 input LUT? [and2.zip](https://github.com/YosysHQ/yosys/files/7707364/and2.zip) here is the same design as before, but in the gate design I changed the 2 input LUT to a 6 input LUT. I get the following error:

```
Found 2 $equiv cells in equiv:
  Of those cells 1 are proven and 1 are unproven.
  Unproven $equiv $auto$equiv_make.cc:359:find_same_cells$1738: \o_OBUF_gold \o_OBUF_gate
Found a total of 1 unproven $equiv cells.
ERROR: Found 1 unproven $equiv cells in 'equiv_status -assert'.
```

Do you know why this is? What would be the solution to this problem?

TheDHCreator, 2022-01-13:
@clairexen I am now working on this issue as well. I replicated the same error that Jarom was having where the equivalence between a 6-LUT and a 2-LUT results in 1 unproven cell. What do you think could be causing this issue?

nakengelhardt, 2022-01-14:
`equiv_struct` proves structural equivalence, i.e. that the same signals are connected to the same cell types. The problem above was that the two input designs didn't have the same passes applied to them so they ended up looking different. Unless I am misunderstanding something, a 6-LUT and a 2-LUT would not be structurally equivalent by definition...

nakengelhardt, 2022-01-14:
Sorry, I had missed the post from the 13th, but now I had a look at the example code.

The two are not structurally equivalent, but the other equiv_* passes will also not work directly because yosys also has no clue what functions a "LUT6" cell or a "LUT2" cell implements, as they are considered blackboxes when read with `-lib`. But the simulation models for these cells are synthesizable, so if you remove the `-lib` parameter it will read in the cell contents for the LUTs, and `equiv_simple` can prove them equivalent:

```
read_verilog +/xilinx/cells_sim.v
read_verilog and2_impl.v
prep -flatten
hierarchy -auto-top
rename -top gold
splitnets -ports;;
design -stash gold

read_verilog +/xilinx/cells_sim.v
read_verilog and2_reversed.v
prep -flatten
hierarchy -auto-top
rename -top gate
splitnets -ports;;
design -stash gate

design -copy-from gold -as gold gold
design -copy-from gate -as gate gate
equiv_make gold gate equiv
hierarchy -top equiv;;
equiv_struct;;
equiv_simple
equiv_status -assert
```

TheDHCreator, 2022-01-14:
@nakengelhardt Thank you for that. I have a few basic designs showing up as equivalent now with the script you provided above. I have a 4-bit adder with a register that currently has 4 unproven cells after running the compare.ys script.

I also have that same 4-bit adder with a register with an additional reset signal. After running the compare.ys script, it also currently has 4 unproven cells. Do you think the addition of the logic being in a register or the addition of a reset signal could be messing with the script? How should I go about fixing this?

[add4_r.zip](https://github.com/YosysHQ/yosys/files/7872031/add4_r.zip)
[add4_r_rst.zip](https://github.com/YosysHQ/yosys/files/7872032/add4_r_rst.zip)

TheDHCreator, 2022-01-24:
Have either of you looked at the designs above yet? If you could, please try to do so soon. I know you are busy, but I would really appreciate it. @clairexen @nakengelhardt

clairexen, 2022-01-31:
Since this has a feedback loop that isn't broken by the equivalence points found by comparing names, you'd need `equiv_induct` to prove sequential equivalence.

The following script works for me. (However, with `equiv_induct` there's currently the possibility of false positives.)

```
read_verilog +/xilinx/cells_sim.v
read_verilog add4_r_impl.v
prep -flatten
hierarchy -auto-top
rename -top gold
splitnets -ports;;
design -stash gold
read_verilog +/xilinx/cells_sim.v
read_verilog add4_r_reversed.v
prep -flatten
hierarchy -auto-top
rename -top gate
splitnets -ports;;
design -stash gate
design -copy-from gold -as gold gold
design -copy-from gate -as gate gate
equiv_make gold gate equiv
hierarchy -top equiv;;
opt_clean -purge
show
equiv_simple -seq 10
equiv_induct -seq 10
equiv_status -assert
```

TheDHCreator, 2022-01-31:
@clairexen @nakengelhardt  I got equivalence proven by running the script above as you said, but my purpose in asking about these scripts these last few weeks is for proving equivalence for many different designs. Is there a way to modify the current script above, to make it include equiv_struct, and to have it work on a multitude of designs? Adders of various sizes (with or without registers/reset signals), simple AND gates (with or without registers), Block RAM, and more?

We are trying to get away from using equiv_induct because of the false-positive possibilities (and it isn't the strongest equivalence checking Yosys command, so I've been told. Is that true?)

Please get back to me as soon as you can. Thanks!

TheDHCreator, 2022-02-14:
@nakengelhardt @clairexen Can you take a look at my message above? Thanks!

closed 2022-02-19 by clairexen