YosysHQ/yosys #3078: Mixed inout pin usage leads to off-by-one pin indexing error

gergoerdi, 2021-11-11:
## Steps to reproduce the issue

Verilog source:

```
module Top(
           output [3:0] LED,
           inout [10:1] JA
           );
   
   wire [3:0]           ROWS = JA[10:7];  
   wire [3:0]           COLS = 4'b1110;
   
   assign LED = ROWS;
   assign JA[4:1] = COLS;   
   
endmodule
```

I have a keypad connected to the `JA` port, so some of the bits in the `JA` bus are output (to select the scanned columns) and some are input (to read in the currently selected column from all rows).

## Expected behavior

The state of the selected (0-th) column should show up on the LEDs.

## Actual behavior

If we look at the Yosys synthesis result, we can see that `COLS` gets assigned to `JA[3:0]` instead of `JA[4:1]`. Similarly, the pins read from `JA` are off by one, so instead of `10:7`, `ROWS` (and consequently, `LED`) gets assigned from `9:6`:

```
$ echo 'read_verilog Top.v; synth; show;' | yosys
```
[image]

mwkmwkmwk, 2021-11-11:
This is not an actual error, it's just idiosyncracy of `show` command (it always uses internal yosys bit numbering, which is from 0 LSB).

When written to JSON or other output format, the relevant wire should have the right annotations to be handled correctly

gergoerdi, 2021-11-11:
> This is not an actual error, it's just idiosyncracy of `show` command (it always uses internal yosys bit numbering, which is from 0 LSB).
> 
> When written to JSON or other output format, the relevant wire should have the right annotations to be handled correctly

I'm not convinced. When I synthesize the whole circuit end-to-end, targeting a Xilinx 7 dev board, the only way I can get it to work is if I shift the definition of `COLS` by one, i.e. if I set `COLS = 4'b1101`, and then I can observe the 0th column (!) being scanned. Similarly, the input as read from `ROWS` is then shown off-by-one on the LEDs.

mwkmwkmwk, 2021-11-11:
Then please give an instruction for how to reproduce the problem end-to-end

gergoerdi, 2021-11-11:
Sure, here it is: https://github.com/gergoerdi/yosys-issue-3078/tree/1eb7b9a88b678087e7ace0e3ade7c45e13e39c34

Build with `make`, then upload to a Nexys A7-50T development board with `make upload`. Connect a Digilent 4x4 keypad to the JA user port, and then observe the LED behaviour.

gergoerdi, 2021-11-12:
I have a much simplified version at https://github.com/gergoerdi/yosys-issue-3078/tree/1eb7b9a88b678087e7ace0e3ade7c45e13e39c34 that doesn't require any external peripherals:

```
module Top(
           output [5:2] LED
           );

   assign LED[5:2] = 4'b1101;   
endmodule
```

With Vivado this lights up LEDs 2, 4 and 5; with Symbiflow using Yosys, it lights up only LED 2.

```
$ echo 'read_verilog Top.v; synth_xilinx; show;' | yosys
```

[image]

gatecat, 2021-11-12:
I've got no idea what all the Symbiflow bits are doing, but running `yosys -p "synth_xilinx -top Top -blif /dev/stdout" Top.v` produces:

```blif
.model Top
.inputs
.outputs LED[2] LED[3] LED[4] LED[5]
.names $false
.names $true
1
.names $undef
.subckt OBUF I=$true O=LED[2]
.subckt OBUF I=$false O=LED[3]
.subckt OBUF I=$true O=LED[4]
.subckt OBUF I=$true O=LED[5]
.end
```

which looks like correct BLIF output to me

gergoerdi, 2021-11-12:
Thanks, I've reported this to the Symbiflow project as well.

gatecat, 2021-11-12:
I'd be curious what the `eblif` file from `symbiflow_synth` looks like if you could post it here (I don't have all their wrappers set up)

gergoerdi, 2021-11-12:
> I'd be curious what the `eblif` file from `symbiflow_synth` looks like if you could post it here 

```
# Generated by Yosys 0.9+4270 (git sha1 539d4ee9, x86_64-conda_cos6-linux-gnu-gcc 1.24.0.133_b0863d8_dirty -fvisibility-inlines-hidden -fmessage-length=0 -march=nocona -mtune=haswell -ftree-vectorize -fPIC -fstack-protector-strong -fno-plt -O2 -ffunction-sections -fdebug-prefix-map=/home/runner/work/conda-eda/conda-eda/workdir/conda-env/conda-bld/yosys_1628927583200/work=/usr/local/src/conda/yosys-0.9_5567_g539d4ee9 -fdebug-prefix-map=/home/cactus/.conda/envs/xc7=/usr/local/src/conda-prefix -fPIC -Os -fno-merge-constants)

.model Top
.inputs
.outputs LED[2] LED[3] LED[4] LED[5]
.subckt GND GND=$false
.subckt VCC VCC=$true
.subckt VCC VCC=$undef
.subckt T_INV TI=$true TO=$iopadmap$Top.LED.t
.cname $iopadmap$Top.LED.genblk1.t_inv
.attr module_not_derived 00000000000000000000000000000001
.attr src "/home/cactus/prog/fpga/symbiflow/symbiflow-examples/xc7/install/share/symbiflow/techmaps/xc7_vpr/techmap/cells_map.v:1921.5-1925.4|/home/cactus/prog/fpga/symbiflow/symbiflow-examples/xc7/install/share/symbiflow/techmaps/xc7_vpr/techmap/cells_map.v:1950.11-1953.6"
.subckt OBUFT_VPR I=$true O=LED[2] T=$iopadmap$Top.LED.t
.cname $iopadmap$Top.LED.obuft
.attr module_not_derived 00000000000000000000000000000001
.attr src "/home/cactus/prog/fpga/symbiflow/symbiflow-examples/xc7/install/share/symbiflow/techmaps/xc7_vpr/techmap/cells_map.v:1921.5-1925.4|/home/cactus/prog/fpga/symbiflow/symbiflow-examples/xc7/install/share/symbiflow/techmaps/xc7_vpr/techmap/cells_map.v:2064.5-2068.4"
.param DRIVE 00000000000000000000000000001100
.param IOSTANDARD "LVCMOS33"
.param IO_LOC_PAIRS "NONE"
.param LVCMOS12_DRIVE_I12 0
.param LVCMOS12_DRIVE_I4 0
.param LVCMOS12_LVCMOS15_LVCMOS18_LVCMOS25_LVCMOS33_LVTTL_SLEW_FAST 0
.param LVCMOS12_LVCMOS15_LVCMOS18_LVCMOS25_LVCMOS33_LVTTL_SSTL135_SSTL15_SLEW_SLOW 1
.param LVCMOS12_LVCMOS15_LVCMOS18_SSTL135_SSTL15_STEPDOWN 0
.param LVCMOS12_LVCMOS25_DRIVE_I8 0
.param LVCMOS15_DRIVE_I12 0
.param LVCMOS15_DRIVE_I8 0
.param LVCMOS15_LVCMOS18_LVCMOS25_DRIVE_I4 0
.param LVCMOS15_SSTL15_DRIVE_I16_I_FIXED 0
.param LVCMOS18_DRIVE_I12_I8 0
.param LVCMOS18_DRIVE_I16 0
.param LVCMOS18_DRIVE_I24 0
.param LVCMOS25_DRIVE_I12 0
.param LVCMOS25_DRIVE_I16 0
.param LVCMOS33_DRIVE_I16 0
.param LVCMOS33_LVTTL_DRIVE_I12_I16 1
.param LVCMOS33_LVTTL_DRIVE_I12_I8 0
.param LVCMOS33_LVTTL_DRIVE_I4 0
.param LVTTL_DRIVE_I24 0
.param PULLTYPE "NONE"
.param PULLTYPE_KEEPER 0
.param PULLTYPE_NONE 1
.param PULLTYPE_PULLDOWN 0
.param PULLTYPE_PULLUP 0
.param SLEW "SLOW"
.param SSTL135_DRIVE_I_FIXED 0
.param SSTL135_SSTL15_SLEW_FAST 0
.subckt T_INV TI=$true TO=$iopadmap$Top.LED_1.t
.cname $iopadmap$Top.LED_1.genblk1.t_inv
.attr module_not_derived 00000000000000000000000000000001
.attr src "/home/cactus/prog/fpga/symbiflow/symbiflow-examples/xc7/install/share/symbiflow/techmaps/xc7_vpr/techmap/cells_map.v:1921.5-1925.4|/home/cactus/prog/fpga/symbiflow/symbiflow-examples/xc7/install/share/symbiflow/techmaps/xc7_vpr/techmap/cells_map.v:1950.11-1953.6"
.subckt OBUFT_VPR I=$false O=LED[3] T=$iopadmap$Top.LED_1.t
.cname $iopadmap$Top.LED_1.obuft
.attr module_not_derived 00000000000000000000000000000001
.attr src "/home/cactus/prog/fpga/symbiflow/symbiflow-examples/xc7/install/share/symbiflow/techmaps/xc7_vpr/techmap/cells_map.v:1921.5-1925.4|/home/cactus/prog/fpga/symbiflow/symbiflow-examples/xc7/install/share/symbiflow/techmaps/xc7_vpr/techmap/cells_map.v:2064.5-2068.4"
.param DRIVE 00000000000000000000000000001100
.param IOSTANDARD "LVCMOS33"
.param IO_LOC_PAIRS "NONE"
.param LVCMOS12_DRIVE_I12 0
.param LVCMOS12_DRIVE_I4 0
.param LVCMOS12_LVCMOS15_LVCMOS18_LVCMOS25_LVCMOS33_LVTTL_SLEW_FAST 0
.param LVCMOS12_LVCMOS15_LVCMOS18_LVCMOS25_LVCMOS33_LVTTL_SSTL135_SSTL15_SLEW_SLOW 1
.param LVCMOS12_LVCMOS15_LVCMOS18_SSTL135_SSTL15_STEPDOWN 0
.param LVCMOS12_LVCMOS25_DRIVE_I8 0
.param LVCMOS15_DRIVE_I12 0
.param LVCMOS15_DRIVE_I8 0
.param LVCMOS15_LVCMOS18_LVCMOS25_DRIVE_I4 0
[... 74 lines cut ...]
.param LVCMOS12_LVCMOS25_DRIVE_I8 0
.param LVCMOS15_DRIVE_I12 0
.param LVCMOS15_DRIVE_I8 0
.param LVCMOS15_LVCMOS18_LVCMOS25_DRIVE_I4 0
.param LVCMOS15_SSTL15_DRIVE_I16_I_FIXED 0
.param LVCMOS18_DRIVE_I12_I8 0
.param LVCMOS18_DRIVE_I16 0
.param LVCMOS18_DRIVE_I24 0
.param LVCMOS25_DRIVE_I12 0
.param LVCMOS25_DRIVE_I16 0
.param LVCMOS33_DRIVE_I16 0
.param LVCMOS33_LVTTL_DRIVE_I12_I16 1
.param LVCMOS33_LVTTL_DRIVE_I12_I8 0
.param LVCMOS33_LVTTL_DRIVE_I4 0
.param LVTTL_DRIVE_I24 0
.param PULLTYPE "NONE"
.param PULLTYPE_KEEPER 0
.param PULLTYPE_NONE 1
.param PULLTYPE_PULLDOWN 0
.param PULLTYPE_PULLUP 0
.param SLEW "SLOW"
.param SSTL135_DRIVE_I_FIXED 0
.param SSTL135_SSTL15_SLEW_FAST 0
.end
```

mwkmwkmwk, 2021-11-12:
I'll take care of the `show` issue; otherwise, this is a symbiflow bug

PR #3079 (2021-11-12): show: Fix wire bit indexing.
Fixes #3078.
commit 107aad2c (2021-11-12): show: Fix wire bit indexing.

Fixes #3078.
closed 2021-11-12 by mwkmwkmwk