YosysHQ/yosys #3092: Yosys ABC error/segmentation fault in SymbiFlow

regymm, 2021-11-29:
Hope this is the place for yosys issue in SymbiFlow as well. 

## Steps to reproduce the issue

I'm using SymbiFlow for xc7z010 following [this](https://symbiflow-examples.readthedocs.io/en/latest/getting-symbiflow.html) installation and Makefile same as [in the SymbiFlow examples](https://github.com/SymbiFlow/symbiflow-examples/blob/master/common/common.mk), `symbiflow_synth -t ${TOP} -v ${VERILOG} -d ${BITSTREAM_DEVICE} -p ${PARTNAME} -x ${XDC}`. 

Symbiflow_synth reported segmentation fault or similar errors for this piece of code, which is trimmed to as short as I can from a CPU cache design that can be run in Vivado with no problem: 

```verilog
`timescale 1ns / 1ps

module cache_cpu
    (
        input clk,

        input [31:0]a,
        input [31:0]d,
        input we,
	output ready
    );

	localparam IDLE = 1;
	localparam LOAD = 3;
	reg [3:0]state = IDLE;

	assign ready = !(we) & (state == IDLE);

	reg way_en;
	reg way_tag_we;
    always @ (*) begin
		way_en = 0;
		way_tag_we = 0;
		case (state)
			IDLE: begin
				if (way_tag_out) begin
					way_tag_we = 1;
				end
			end
			LOAD: begin
				way_en = 1;
				way_tag_we = 1;
			end
		endcase
    end

    always @ (posedge clk) begin
		case(state)
			IDLE: begin
				if (way_tag_out) begin
					state <= LOAD;
				end
			end
			LOAD: begin
				state <= IDLE;
			end
		endcase
    end

	wire [31:0]way_tag_out;

	cacheway way_gen (
		.clk(clk),
		.en(way_en),
		.tag_we(way_tag_we),
		.tag_out(way_tag_out),
	);

endmodule
module cacheway
    (
        input clk,
		input en,
		input tag_we,
		output [31:0]tag_out
    );

	reg [31:0]tags[31:0];

	reg state = 0;
	reg [9:0]count = 0;
	always @ (posedge clk) begin
		if (state == 0) begin
			count <= count + 1;
			if (count == 10) state <= 1;
			tags[count] <= 0;
		end else
			if (en & tag_we) tags[0] <= 23;
			//if (tag_we) tags[0] <= 23; // and remove all en
	end
	assign tag_out = tags[0];

endmodule
```

The whole compile output is [here](https://gist.github.com/regymm/3cfaefb1cc5922d72a8d1ccc775efddf). 

The yosys version is managed by SymbiFlow installer so I'm not sure it's git master or not.

Ravenslofty, 2021-11-29:
> The yosys version is managed by SymbiFlow installer so I'm not sure it's git master or not.

`Yosys 0.9+3962` is 5d0cc54f from way back in February, and there have been [multiple ABC9 bug fixes](https://github.com/YosysHQ/yosys/pulls?q=is%3Apr+is%3Aclosed+created%3A>%3D2021-02-25+abc9) since then.

I can't reproduce this with 707d98b0.

regymm, 2021-11-29:
Oh, seems my SymbiFlow toolchain is too old. After bumping to latest this is solved.

closed 2021-11-29 by regymm