YosysHQ/yosys #3093: Latches on conditions optimized as X upon multiple conditions

suzizecat, 2021-11-29:
## Steps to reproduce the issue
On the current master (707d98b06c8d2ad196ba64afbbb3bf5af475c75b) , on Ubuntu 20.04,

With the following *SystemVerilog* code : 
```systemverilog
module mwe (
	input  logic       i_clk   , //!
	input  logic       i_rst_n , //!
	input  logic       i_valid , //!
	input  logic       i_force , //!
	output logic [2:0] o_signal  //!
);

	always_ff @(posedge i_clk or negedge i_rst_n) begin : p_seq
		if (~ i_rst_n ) begin
			o_signal <= 0;
		end else begin
			if (i_force) begin
				o_signal <= 0;
			end else if (i_valid) begin
				o_signal <= o_signal +1;
			end
		end
	end

endmodule
```
and using yosys with the command `read_verilog -sv mwe.sv ; proc ; opt ; show` I get the following graph, on which you can notice the `3'x` that I highlighted.

[image]

## Expected behavior

I would expect the `$4 $mux` to have a latched `o_signal` value instead of the `3'x`  

## Actual behavior

`$4 $mux` have a `X` value at its input which will probably be optimized out later.

This seems to be due to the fact that `o_signal` is in two different conditions as having only the following code in the "active" (not under reset) region of `p_seq` works as expected. 
```systemverilog
if (i_valid) begin
    o_signal <= o_signal +1;
end
```
Works as expected.

A workaround, in this situation can be to use the following code, which works as expected. 
```systemverilog
if (i_force | i_valid) begin
    o_signal <= i_force ? 0 : o_signal +1;
end
```
And produce this graph: 
[image]
 

FWIW, this issue also reproduce using a Surelog + UHDM + Yosys setup.

Best regards
Julien FAUCHER

mwkmwkmwk, 2021-11-30:
There is no bug here — the input to the multiplexer for this case doesn't matter, as the FF is prevented from changing via its EN input when this input is active.

closed 2021-11-30 by mwkmwkmwk

suzizecat, 2021-12-01:
You indeed have a point. 
The issue is that it seems that Yosys optimize way too much things away later in synthesis (during the abc pass if I remember correctly) so there still kind of an issue, maybe not where I though, I'll admit.

suzizecat, 2021-12-01:
After giving it a closer look, I actually don't get your point.
I may have missed a point, but if I have `i_valid` with the value that select A (hence X) it will always propagate X to the flop, regardless of the enable (and therefore, X will be propagated with enable on the flop).
As you can see, the enable condition is actually different from mux `$4`  selection condition so the issue stands. 

For **this particular case** to work, Yosys could have simply removed `$4` altogether. This might not be the case with more complex situations though.

In order to be equivalent to my code, `$4.A` should rather have `$10.Q` as source instead of X value, which would be a proper latch. (I'm not considering the actual, physical, behavior of such structure though)