YosysHQ/yosys #3101: MacOS: Stripping breaks binaries

tristanitschner, 2021-12-10:
## Steps to reproduce the issue

run sudo make install && yosys on mac

*Provide instructions for reproducing the issue. Make sure to include
all necessary source files. (You can simply drag&drop a .zip file into
the issue editor.)*

Also, make sure that the issue is actually reproducable in current git
master of Yosys.

See https://stackoverflow.com/help/mcve for some information on how to
create a Minimal, Complete, and Verifiable example (MCVE).

Please do not waste our time with issues that lack sufficient information
to reproduce the issue easily. We will simply close those issues.

Contact https://www.yosyshq.com/ if you need commercial support for Yosys.

## Expected behavior

yosys should start yosys

*Please describe the behavior you would have expected from the tool.*

## Actual behavior

shell returns sigkill

*Please describe how the behavior you see differs from the expected behavior.*

1. the owner of the files /usr/local/bin/yosis* shouldn't be root
2. the stripping of the binaries in the sudo make install somehow breaks them (GNU strip (GNU Binutils) 2.33.1)
3. if the binaries are manually copied to /usr/local/bin without stripping, all works fine

kammoh, 2021-12-10:
Hi,
I'm also seeing this issue on macOS/arm64 Monterey: after running 'make install' for a second time (or on a system where yosys binary already exists in the target path), the executable is killed immediately. It's not even possible to run the installed executable from lldb! The process is killed immediately, after launch.

The issue is not really coming from stripping. There's nothing wrong with the stripped binaries, you can try copying the unstripped binary to the installation location and the same thing happens. Also, if I run either unstripped or manually stripped executable from the build folder, it works without any problem. 

Looking at the crash report from "console.app" sheds some light:
```
_dyld_process_info_create failed with 6
dyld_process_snapshot_get_shared_cache failed
Failed to create CSSymbolicatorRef - corpse still valid ¯\_(ツ)_/¯
```
and in the "Full Report":
```
"type":"EXC_BAD_ACCESS","signal":"SIGKILL (Code Signature Invalid)","subtype":"UNKNOWN_0x32 at 0x00000001040a0000"},
```
Aha!
But the yosys executable is not signed at all.
`dyld_process_snapshot_get_shared_cache` suggest some sort of bug in the loader cache.

tristanitschner, 2021-12-10:
I'm sorry, but I have to correct you.
Running ./yosys* from the repository's root works.
Copying yosys* to /usr/local/bin works aswell. (This is my current setup.)
Pls run strip -S yosis one some copy of the executable. It won't run wherever you place it.
However, a quick google search turned up:
https://developer.apple.com/forums/thread/666022
I postulate apple #!**** something up and they don't really care (that thread is from more than a year ago..., save for the meaningful error message)
I just hope someone else who stumbles upon this problem will find this fix.

kammoh, 2021-12-15:
I described the behavior I saw on my machine, maybe it behaved differently on yours.
The issue is fixed in macOS 12.1, so I have no ways to reproduce it.

Anyways I think this issue can be closed.
It was a macOS bug, and it's now fixed!