YosysHQ/yosys #3112: CXXRTL: internal assertion on reduced testcase

Wren6991, 2021-12-15:
## Steps to reproduce the issue

Repro files: [hazard3-alu-cxxrtl-repro.zip](https://github.com/YosysHQ/yosys/files/7717168/hazard3-alu-cxxrtl-repro.zip)

This zip file contains two directories. Each directory is self-contained, with source and `.ys` files.

* `verilog_src` -- part of the Verilog hierarchy of my project, bisected as far as possible (the ALU and two subordinate modules)
* `ilang_reduced` -- the above, reduced with bugpoint _as far as possible_. I had some issues here with bugpoint not terminating

`cd` into either directory, and run 

```
yosys bug.ys
```

I'm on `Yosys 0.12+23 (git sha1 5e5c8a54c, clang 10.0.0-4ubuntu1 -fPIC -Os)`, which was `master` on Monday. You should see:

```
ERROR: Assert '!is_lhs' failed in backends/cxxrtl/cxxrtl_backend.cc:965.
```

## Expected behavior

Produce a C++ output, or a diagnostic saying an issue with the Verilog that I can fix

## Actual behavior

```
ERROR: Assert '!is_lhs' failed in backends/cxxrtl/cxxrtl_backend.cc:965.
```

## Further Troubleshooting

I was previously on Yosys version `c2afcbe7`, which did not trip this same assertion, and as far as I can tell produced working build output. I have gone back to rebuild that version of Yosys, and run it on the repro zip above, and it does not trip this assertion.

`c2afcbe7` was a few commits after the fix for #2882, which was in fact the same assertion tripping on an earlier version of this same ALU.

I tried to reduce the testcase from the full source of my project, and found that bugpoint didn't terminate, even when left running overnight. Eventually Yosys was killed. After I bisected down to a smaller part of my project source, I _still_ found that bugpoint would not terminate. It continually "finds something". In the reduced testcase above I see a lot of wires called `$delete_wire$xxxx` and I wonder if bugpoint is somehow looping on deleting its own created wires. I'll open another issue for this.

I reduced as far as I could by hacking `bugpoint` to output its last successful testcase after hitting 20 000 iterations.

whitequark, 2021-12-15:
Further reduced:
```
autoidx 10151
module \hazard3_alu
  wire width 32 $0\result[31:0]
  wire width 32 $1\result[31:0]
  wire input 2 $auto$bugpoint.cc:258:simplify_something$10150
  wire output 1 $auto$bugpoint.cc:258:simplify_something$746
  wire $delete_wire$10143
  wire width 6 $delete_wire$752
  wire $ternary$hazard3_alu.v:58$55_Y
  wire \lt
  wire width 32 \result
  cell $mux $ternary$hazard3_alu.v:58$55
    parameter \WIDTH 1
    connect \A 1'x
    connect \B $auto$bugpoint.cc:258:simplify_something$10150
    connect \S $delete_wire$10143
    connect \Y $ternary$hazard3_alu.v:58$55_Y
  end
  cell $mux $ternary$hazard3_alu.v:62$58
    parameter \WIDTH 1
    connect \A \lt
    connect \B 1'x
    connect \S 1'x
    connect \Y $auto$bugpoint.cc:258:simplify_something$746
  end
  process $proc$hazard3_alu.v:148$213
    assign $0\result[31:0] $1\result[31:0]
    switch { 4'0000 $delete_wire$752 }
      case 10'----000100
        assign $1\result[31:0] { 31'0000000000000000000000000000000 \lt }
    end
    sync always
      update \result $0\result[31:0]
  end
  connect \lt $ternary$hazard3_alu.v:58$55_Y
end
```

Wren6991, 2021-12-15:
Another testcase which trips this assertion, reduced from the full processor source thanks to #3114:

```
# Testcase 2, reduced from full processor source

module \hazard3_decode
  wire width 4 $6\d_except[3:0]
  attribute $bugpoint 1
  wire width 2 $auto$bugpoint.cc:258:simplify_something$9582
  attribute $bugpoint 1
  wire width 2 output 1 $auto$bugpoint.cc:258:simplify_something$9584
  attribute $bugpoint 1
  wire width 4 $auto$bugpoint.cc:258:simplify_something$9589
  attribute $bugpoint 1
  wire $auto$bugpoint.cc:258:simplify_something$9592
  wire $logic_or$/home/luke/proj/hazard3/hdl/hazard3_decode.v:298$1981_Y
  wire width 4 $procmux$2920_Y
  wire $procmux$2921_CMP
  wire width 4 $procmux$2923_Y
  wire $procmux$2924_CMP
  wire $procmux$2936_CMP
  wire width 4 \d_except
  cell $logic_or $logic_or$/home/luke/proj/hazard3/hdl/hazard3_decode.v:298$1981
    parameter \A_SIGNED 0
    parameter \A_WIDTH 1
    parameter \B_SIGNED 0
    parameter \B_WIDTH 1
    parameter \Y_WIDTH 1
    connect \A 1'x
    connect \B $auto$bugpoint.cc:258:simplify_something$9592
    connect \Y $logic_or$/home/luke/proj/hazard3/hdl/hazard3_decode.v:298$1981_Y
  end
  attribute \full_case 1
  cell $mux $procmux$2920
    parameter \WIDTH 4
    connect \A 4'x
    connect \B $auto$bugpoint.cc:258:simplify_something$9589
    connect \S $procmux$2921_CMP
    connect \Y $procmux$2920_Y
  end
  attribute \full_case 1
  cell $mux $procmux$2923
    parameter \WIDTH 4
    connect \A 4'x
    connect \B $6\d_except[3:0]
    connect \S $procmux$2924_CMP
    connect \Y $procmux$2923_Y
  end
  attribute \full_case 1
  cell $mux $procmux$2935
    parameter \WIDTH 2
    connect \A 2'x
    connect \B $auto$bugpoint.cc:258:simplify_something$9582
    connect \S $procmux$2936_CMP
    connect \Y $auto$bugpoint.cc:258:simplify_something$9584
  end
  cell $paramod\hazard3_instr_decompress\PASSTHROUGH=1'1 \decomp
    connect \instr_in 32'x
    connect \instr_is_32bit 1'x
    connect \instr_out 32'x
    connect \invalid 1'x
  end
  connect $procmux$2921_CMP $logic_or$/home/luke/proj/hazard3/hdl/hazard3_decode.v:298$1981_Y
  connect $6\d_except[3:0] $procmux$2920_Y
  connect $procmux$2924_CMP $logic_or$/home/luke/proj/hazard3/hdl/hazard3_decode.v:298$1981_Y
  connect $procmux$2936_CMP $logic_or$/home/luke/proj/hazard3/hdl/hazard3_decode.v:298$1981_Y
  connect \d_except $procmux$2923_Y
end
attribute \hdlname "\\hazard3_instr_decompress"
attribute \dynports 1
module $paramod\hazard3_instr_decompress\PASSTHROUGH=1'1
  parameter \PASSTHROUGH 1'1
end
```

whitequark, 2021-12-15:
#3115

Wren6991, 2021-12-15:
#3115 fixes the assertion, I'll see if it sims

Wren6991, 2021-12-15:
Yep, I can run software testcases on the processor. Thanks!

PR #3115 (2021-12-16): cxxrtl: demote wires not inlinable only in debug_eval to locals
Fixes #3112.
commit 7f2ea7d2 (2021-12-16): cxxrtl: demote wires not inlinable only in debug_eval to locals.

Fixes #3112.
closed 2021-12-16 by whitequark