YosysHQ/yosys #3113: Bugpoint does not terminate when reducing testcase

Wren6991, 2021-12-15:
## Steps to reproduce the issue

Meta-issue for #3112. Repro files:

[hazard3-bugpoint-loop.zip](https://github.com/YosysHQ/yosys/files/7717395/hazard3-bugpoint-loop.zip)

The zip contains a `.il` file with the testcase from #3112, and a script that tries to reduce this testcase using bugpoint. After extracting the zip, run:

```
yosys bug.ys
```

Yosys version I used (latest `master`):

```
Yosys 0.12+30 (git sha1 477eeefd9, clang 10.0.0-4ubuntu1 -fPIC -Os)
```

## Expected behavior

Bugpoint eventually terminates, and produces a minimised testcase which can be further reduced by hand

## Actual behavior

Bugpoint does not terminate. I tried leaving it running overnight, and Yosys was eventually killed due to OOM.

whitequark, 2021-12-15:
#3114

Wren6991, 2021-12-15:
Thanks, will try.

Wren6991, 2021-12-15:
Yep, #3114 allows me to reduce a testcase from the full processor source, as well as the repro here. Thanks!

PR #3114 (2021-12-16): bugpoint: avoid infinite loop between -connections and -wires
Fixes #3113.
commit 4f1d62d9 (2021-12-16): bugpoint: avoid infinite loop between -connections and -wires.

Fixes #3113.
closed 2021-12-15 by Wren6991