YosysHQ/yosys #3117: "ERROR: Assert `port.addr[j] == State::S0' failed" when in Verific mode

emamajanyan, 2021-12-16:
## Steps to reproduce the issue

Use the 'shared_ports' test-case from YosysHQ tests 'tests/memories/shared_ports.v' (https://github.com/YosysHQ/yosys/blob/master/tests/memories/shared_ports.v)

```
read -sv shared_ports.v                                                             
hierarchy -check -top test
proc
opt_clean
memory -nomap
```

Yosys 0.12+30 (git sha1 ed4642e18, clang 10.0.0-4ubuntu1 -fPIC -Os)

## Expected behavior

The test should run without any assertion like in case with 'read_verilog' command:

```
read_verilog -sv shared_ports.v                                                             
hierarchy -check -top test
proc
opt_clean
memory -nomap
```

## Actual behavior

The test fails during MEMORY_SHARE pass with the following assertion.
ERROR: Assert `port.addr[j] == State::S0' failed in kernel/mem.cc:488.

clairexen, 2021-12-20:
Please contact support@yosyshq.com for YosysHQ’s commercial support, since this concerns an issue involving non-open-source components only included in Tabby CAD Suite.

mwkmwkmwk, 2021-12-20:
this is a `memory_share` bug, reproducible without Verific:

```
module test (...);

input [7:1] wa1;
input [7:1] wa2;
input [7:0] ra;
output [7:0] rd;
input clk;
input we1, we2;
input [15:0] wd1, wd2;

reg [7:0] mem [0:255];

assign rd = mem[ra];

always @(posedge clk) begin
        if (we1) begin
                mem[{wa1, 1'b0}] <= wd1[7:0];
                mem[{wa1, 1'b1}] <= wd1[15:8];
        end else begin
                mem[{wa2, 1'b0}] <= wd2[7:0];
                mem[{wa2, 1'b1}] <= wd2[15:8];
        end
end

endmodule
```

`yosys -p 'hierachy -top test; proc; opt; memory_share'`

PR #3123 (2021-12-20): memory_share: Fix SAT-based sharing for wide ports.
Fixes #3117.
commit f84c9d8e (2021-12-20): memory_share: Fix SAT-based sharing for wide ports.

Fixes #3117.
closed 2021-12-20 by mwkmwkmwk