YosysHQ/yosys #3121: Crash in `proc_dlatch`

modwizcode, 2021-12-18:
Reduced manually.

MCVE:
command:
`yosys -p "read_rtlil crash.il; proc_dlatch"`

crash.il:
```
module \top
  cell $mux $2
  end
end
```

log:
```
 Yosys 0.12+30 (git sha1 ed4642e18, ccache clang 13.0.0 -fPIC -Os)

-- Running command `read_rtlil crash.il; proc_dlatch' --

1. Executing RTLIL frontend.
Input filename: crash.il

2. Executing PROC_DLATCH pass (convert process syncs to latches).
terminate called after throwing an instance of 'std::out_of_range'
  what():  dict::at()
```

mwkmwkmwk, 2021-12-18:
This is invalid RTLIL (yosys won't even load it when built in debug mode), because of missing ports/params on internal cell.

closed 2021-12-18 by mwkmwkmwk

modwizcode, 2021-12-18:
We figured as much, but wanted to make sure.