YosysHQ/yosys #3125: Inconsistent handling of multiple default_nettype directives

Wren6991, 2021-12-21:
## Steps to reproduce the issue

Filename `default_nettype.v`:

```verilog
`default_nettype none

module m (
	input  wire a,
	output wire b
);

assign b = x;

endmodule

`default_nettype wire
```

Run:

```bash
yosys -p "read_verilog default_nettype.v"
```

## Expected behavior

An error like:

```
default_nettype.v:8: ERROR: Identifier `\x' is implicitly declared and `default_nettype is set to none.
```

## Actual behavior

No error. An implicitly declared net is created.

## Notes

This pattern is used to avoid breaking other code, which may rely on `default_nettype` being `wire`, since `default_nettype` is global (or at least, has greater than file scope) in many tools.

This might stem from an ambiguity in the language reference (1364-2005 19.2):

[image]

Specifically:
> Multiple  `default_nettype  directives  are  allowed.  The latest occurrence of this directive in the source controls the type of nets that will be implicitly declared

There are two intepretations of "latest":
* Yosys' behaviour seems be "the _last_ `default_nettype` in the entire file prevails". If I append another `default_nettype none` to the repro above, it does raise an error.
* The behaviour of big-iron simulators like Xcelium seems to be  "at any point in the file, the _most recent_ `default_nettype` prevails".

The second interpretation makes it easier to write portable HDL that uses `default_nettype none` internally.

Ravenslofty, 2021-12-21:
Duplicate of #1217.

Wren6991, 2021-12-21:
Sorry, I searched but didn't see that. I'll close this.