YosysHQ/yosys #3131: some question about yosys

kangliyu1, 2021-12-27:
Hello! When using yosys, I found that the results synthesized by the script below are not ideal, so I would like to ask if any seniors can give pointers to the deficiencies of the script below

```
# Yosys synthesis script for ${TOP_MODULE}

#########################
# Parse input files
#########################
# Read verilog files
${READ_VERILOG_FILE}
# Read technology library
read_verilog -lib -specify ${YOSYS_CELL_SIM_VERILOG}

#########################
# Prepare for synthesis
#########################
# Identify top module from hierarchy
hierarchy -check -top ${TOP_MODULE}
# - Convert process blocks to AST
proc
# Flatten all the gates/primitives
flatten
# Identify tri-state buffers from 'z' signal in AST
# with follow-up optimizations to clean up AST
tribuf -logic
opt_expr
opt_clean
# demote inout ports to input or output port
# with follow-up optimizations to clean up AST
deminout
opt

opt_expr
opt_clean
check
opt
wreduce -keepdc
peepopt
pmuxtree
opt_clean

########################
# Map multipliers
# Inspired from synth_xilinx.cc
#########################
# Avoid merging any registers into DSP, reserve memory port registers first
memory_dff
wreduce t:$mul
techmap -map +/mul2dsp.v -map ${YOSYS_DSP_MAP_VERILOG} ${YOSYS_DSP_MAP_PARAMETERS}
select a:mul2dsp
setattr -unset mul2dsp
opt_expr -fine
wreduce
select -clear
chtype -set $mul t:$__soft_mul# Extract arithmetic functions

#########################
# Run coarse synthesis
#########################
# Run a tech map with default library
techmap
alumacc
share
opt
fsm
# Run a quick follow-up optimization to sweep out unused nets/signals
opt -fast
# Optimize any memory cells by merging share-able ports and collecting all the ports belonging to memorcy cells  
memory -nomap
opt_clean

#########################
# Map logics to BRAMs
#########################
memory_bram -rules ${YOSYS_BRAM_MAP_RULES}
techmap -map ${YOSYS_BRAM_MAP_VERILOG}
opt -fast -mux_undef -undriven -fine
memory_map
opt -undriven -fine

#########################
# Map flip-flops
#########################
techmap -map ${YOSYS_DFF_MAP_VERILOG}
opt_expr -mux_undef
simplemap
opt_expr
opt_merge
opt_rmdff
opt_clean
opt

#########################
# Map LUTs
#########################
#abc -lut ${LUT_SIZE}
abc -lut 4:6

#########################
# Check and show statisitics
#########################
hierarchy -check
stat

#########################
# Output netlists
#########################
opt_clean -purge
write_blif ${OUTPUT_BLIF}
```

Ravenslofty, 2021-12-28:
The immediate question I have is "why are you trying to do this".

It's difficult to tell which pass is the most effective, because you haven't sent your testcase.

It's possible, for example, that you have a large ROM which is mostly zeroes; `memory_bram` maps this to initialised memory blocks, but since `synth` doesn't support those, it turns the ROM into flops and muxes, and the optimisation passes discover these can be reduced significantly.

Or, since `synth` implies a call to `abc` as-is, the multiple passes of ABC produce much better logic than the single pass does.

kangliyu1, 2021-12-28:
> The immediate question I have is "why are you trying to do this".
>
> It's difficult to tell which pass is the most effective, because you haven't sent your testcase.
>
> It's possible, for example, that [...]

@Ravenslofty Hello!I wonder if you could give me some advice on this code under Synth?
My testcase:
```
`timescale 1ns / 1ps
//////////////////////////////////////////////////////////////////////////////////
// Company: 
// Engineer: 
// 
// Create Date: 2020/08/01 08:30:12
// Design Name: 
// Module Name: Conv_1
// Project Name: 
// Target Devices: 
// Tool Versions: 
// Description: 
// 
// Dependencies: 
// 
// Revision:
// Revision 0.01 - File Created
// Additional Comments:
// 
//////////////////////////////////////////////////////////////////////////////////

module Conv_1#(
        parameter padding       =      0                                ,
        parameter weight_bit    =      10                               ,//鏈夌鍙蜂綅瀹?
        parameter conv_bit      =      1                               ,//鏃犵鍙蜂綅瀹?
        parameter conv_wid      =      28                               ,
        parameter conv_heig     =      28                               ,
        parameter stride        =      1                                ,
        parameter result_len    =      26*26                            ,
        parameter in_layer      =      1                                ,
        parameter out_layer     =      6                                
)(
        input			            			                                        clk	            ,
        input			            			                                        rst	            ,
        input		    [in_layer*conv_bit-1:0]			                                conv_din	    ,//1
        input                                                                           din_vld         ,
        input                                                                           din_last        ,
        input           [0:2*in_layer*9*weight_bit-1]                                   kernel          ,//2*1*90
        output          [in_layer*out_layer*(4+weight_bit+conv_bit)/1-1:0]              conv_dout       ,//1*6*15
        output                                                                          busy            ,
        output          [clogb2(out_layer)-1:0]                                         kernel_addr     ,//3
        output  reg                                                                     kernel_vld      ,
        output  reg                                                                     dout_vld        ,
        output  reg                                                                     dout_last       

    );
 /*
rst:高电平有效
din_vld：4周期判断信号，则为1
din_last ：784周期判断信号,是则为1
end_din :cnt_din >= conv_wid * conv_heig din>=输入图像的大小
conv_rst：din_last && din_vld && !end_din，不知要干嘛

 */
        localparam rd_init1              =   3'b000       ,
                   rd_init2              =   3'b001       ,
                   rd_init3              =   3'b010       ,
                   rd_conv               =   3'b011       ; 
        reg		signed  [4+weight_bit+conv_bit-1:0]   add_res   [0:in_layer/1-1][0:out_layer-1]                     ;//15add_res   [0:0][0:5]
        reg  	signed  [weight_bit+conv_bit-1:0]     mul_res   [0:in_layer/1-1][0:out_layer-1][0:9-1]              ;//11  mul_res[0:0][0:5][0:8]
        wire    signed  [weight_bit-1:0]              kernel_val[0:in_layer/1*2*9-1]                                ;//10kernel_val[0:1*2*9-1]
[... 484 lines cut ...]
         2'b11:
           count <= count;
      endcase
   end
end

// empty指针
always @(count) begin
   if (count == 2'b00)
      emptyp <= 1'b1;      //count为0时emptyp赋为1
   else
      emptyp <= 1'b0;
end

// fullp指针
always @(count) begin
   if (count == MAX_COUNT)
      fullp <= 1'b1;                     //计数到最大时fullp赋为1
   else
      fullp <= 1'b0;
end

endmodule

```