YosysHQ/yosys #3132: Dockerfile fails to build ("Test: case_expr_extend -> ERROR!")

Trolldemorted, 2021-12-29:
## Steps to reproduce the issue
- Clone this repo
- Try to build the docker image

## Expected behavior
- Docker image builds successfully

## Actual behavior
```
[100%] Building share/sf2/arith_map.v
[100%] Building share/sf2/cells_map.v
[100%] Building share/sf2/cells_sim.v

  Build successful.

mkdir -p /opt/yosys/bin
cp yosys yosys-config yosys-abc yosys-filterlib yosys-smtbmc /opt/yosys/bin
strip -S /opt/yosys/bin/yosys
strip /opt/yosys/bin/yosys-abc
strip /opt/yosys/bin/yosys-filterlib
mkdir -p /opt/yosys/share/yosys
cp -r share/. /opt/yosys/share/yosys/.
cd tests/simple && bash run-test.sh ""
make[1]: Entering directory '/yosys/tests/simple'
+ gcc -Wall -o /yosys/tests/tools/cmp_tbdata /yosys/tests/tools/cmp_tbdata.c
Test: arrays02 -> ok
Test: asgn_binop -> ok
Test: case_expr_extend -> ERROR!
Note: Make sure that 'iverilog' is an up-to-date git checkout of Icarus Verilog.
make[1]: *** [../tools/autotest.mk:10: case_expr_extend.sv] Error 1
make[1]: Leaving directory '/yosys/tests/simple'
make: *** [Makefile:786: test] Error 2
The command '/bin/sh -c cd /yosys  && make  && make install  && make test' returned a non-zero code: 2
ERROR: Service 'yosys' failed to build : Build failed
```

Trolldemorted, 2021-12-29:
If I remove `make test`, yosys exits with:
```
Yosys 0.12+45 (git sha1 UNKNOWN, clang 7.0.1-8+deb10u2 -fPIC -Os)

ERROR: Assert `pass_register.count(pass_name) == 0' failed in kernel/register.cc:629.
```

zachjs, 2021-12-29:
I believe I have a decent idea what's going wrong here. I'll provide a fix tomorrow.

PR #3134 (2022-01-03): fix iverilog compatibility for new case expr tests
The Docker image uses an older version of iverilog (10.2). I confirmed the build works with these changes, and that the tests are still providing the intended coverage. @Trolldemorted, can you double check that this fixes your build? 

Fixes #3132.
closed 2022-01-03 by zachjs