YosysHQ/yosys #3141: yosys incrorrectly infers latches

eugmes, 2022-01-02:
## Steps to reproduce the issue

I was trying to synthesize some VHDL designs using ghdl and yosys and discovered that there were latches inferred when I expected none. I managed to simplify the verilog produced by ghdl to this:

```verilog
module dff
  (input  clk,
   input  d,
   input  load,
   output q);
  wire r;
  wire rin;
  wire n4_o;
  reg n5_o;
  reg n11_q;
  assign q = r;
  assign r = n11_q;
  assign rin = n5_o;
  assign n4_o = load == 1'b1;
  always @*
    case (n4_o)
      1'b1: n5_o <= d;
    endcase
  always @(posedge clk)
    n11_q <= rin;
endmodule
```

This shows that latches are produced:
```shell
% yosys -p "read_verilog dff.v; synth_xilinx;"
...
   Number of cells:                  7
     BUFG                            1
     FDRE                            1
     IBUF                            3
     LDCE                            1
     OBUF                            1
...
```

Note the LDCE used here, in addition to the expected FDRE.

Strangely, this does not produce any latches, and can be used as workaround:

```shell
% yosys -p "read_verilog dff.v; proc_dff; synth_xilinx;"
...
   Number of cells:                  7
     BUFG                            1
     FDRE                            1
     IBUF                            3
     LUT3                            1
     OBUF                            1
...
```

The problem does not seem to be specific to `synth_xilinx`, running `synth_intel` without explicit `proc_dff` results in a crash:
```
2.18. Executing DFFLEGALIZE pass (convert FFs to types supported by the target).
ERROR: FF dff.$auto$ff.cc:262:slice$95 (type $_DLATCH_P_) cannot be legalized: D latches are not supported
```

I'm not very familiar with verilog, so there may be some problems with the code produced by ghdl, but I find it strange that yosys is able to produce expected result when using `proc_dff`.

Xiretza, 2022-01-02:
The verilog code is definitely iffy - without some folding done by `proc_dff`, that's a register attached to the output of a latch instead of a clean flip-flop. What is the VHDL source code for this? Have you considered using https://github.com/ghdl/ghdl-yosys-plugin instead of exportying verilog from ghdl, then re-importing that into yosys?

gatecat, 2022-01-02:
Can confirm that this produces a latch in Vivado; too (note the LDCE):
[image]

eugmes, 2022-01-02:
@Xiretza here is the original vhdl code (architecture `with_case`, I'm trying to use this design method: https://www.gaisler.com/doc/vhdl2proc.pdf, architecture `with_if` does not produce latch):

```
library IEEE;
use IEEE.std_logic_1164.all;

entity dff is
    port (clk : in std_logic; d, load : in std_logic; q : out std_logic);
end entity dff;

architecture with_case of dff is
    signal r, rin : std_logic;
begin
    comb : process (d, load) is
        variable v : std_logic;
    begin
        v := r;
        case load is
            when '1' => v := d;
            when others => null;
        end case;
        rin <= v;

        q <= r;
    end process comb;

    regs : process (clk) is
    begin
        if rising_edge(clk) then
            r <= rin;
        end if;
    end process regs;
end architecture with_case;

architecture with_if of dff is
    signal r, rin : std_logic;
begin
    comb : process (d, load) is
        variable v : std_logic;
    begin
        v := r;
        if load = '1' then
            v := d;
        end if;
        rin <= v;

        q <= r;
    end process comb;

    regs : process (clk) is
    begin
        if rising_edge(clk) then
            r <= rin;
        end if;
    end process regs;
end architecture with_if;
```

I cannot try ghdl plugin right now because it's not easy on macOS.

I've also noticed that even with `proc_dff` the design contain latch made with LUT3.

@gatecat I guess that's a gdhl bug then. Thanks for verifying.

Xiretza, 2022-01-02:
Those are some really weird ways to model a flip-flop - I guess it would be nice if yosys was able to figure out that it's a flip-flop, but I don't think I've ever seen such a construct in the wild.

eugmes, 2022-01-02:
@Xiretza well, I'm not designing a flip-flop, that was a simplified example ;) My still quite small design produced >1000 LUTs :/. Anyway, that's probably a ghdl bug, so I'm closing this one.

closed 2022-01-02 by eugmes