YosysHQ/yosys #3147: About being killed when using yosys synthesis

kangliyu1, 2022-01-05:
Hello! When using script 1 to synthesize code1, there was a problem of being killed in the MEMORY_MAP step of synth. As shown in Figure 1, code1 can be synthesized normally in vivado. I feel very confused about this. I hope someone can give me some pointers. Thank you very much! !

script 1：
read_verilog /home/kly/OpenFPGA/openfpga_flow/tasks/benchmark_sweep/mac_units_2/kly_test/Conv_3.v
hierarchy -top Conv_3
proc
techmap -D NO_LUT -map /home/kly/OpenFPGA/yosys/share/adff2dff.v
synth -top Conv_3 -flatten
clean
abc -lut 6
synth -run check
opt_clean -purge
write_blif Conv_3_    yosys_out.blif

fig1：
[image]

code1:
```
`timescale 1ns / 1ps
module Conv_3#(
        parameter padding       =      0                                ,
        parameter weight_bit    =      10                               ,//鏈夌鍙蜂綅瀹?
        parameter conv_bit      =      10                               ,//鏃犵鍙蜂綅瀹?
        parameter conv_wid      =      10                               ,
        parameter conv_heig     =      10                              ,
        parameter stride        =      1                                ,
        parameter result_len    =      8*8                            ,
        parameter in_layer      =      16                                ,
        parameter out_layer     =      16                                
)(
        input			            			                                        clk	            ,
        input			            			                                        rst	            ,
        input		    [159:0]			                                conv_din	    ,
        input                                                                           din_vld         ,
        input                                                                           din_last        ,
        input           [0:1439]                                   kernel          ,
        output          [3071:0]              conv_dout       ,
        output                                                                          busy            ,
        output          [4:0]                                         kernel_addr     ,
        output  reg                                                                     kernel_vld      ,
        output  reg                                                                     dout_vld        ,
        output  reg                                                                     dout_last       

    );
        localparam rd_init1              =   3'b000       ,
                   rd_init2              =   3'b001       ,
                   rd_init3              =   3'b010       ,
                   rd_conv               =   3'b011       ; 
        reg		signed  [23:0]   add_res   [0:7][0:15]                 ;
        reg  	signed  [19:0]     mul_res   [0:7][0:15][0:9-1]          ;  
        wire    signed  [9:0]              kernel_val[0:71]                            ;
        reg		signed  [10:0]                  conv_buf  [0:2-1][0:7][0:9-1]                  ;        
        wire	        [159:0]	      row_out	[0:3-1]                                           ;       		
        wire	        [159:0]	      row_in	[0:3-1]                                           ;      
        reg		[9:0]                         cnt_row0         		            ; 		
        reg		[9:0]                         cnt_row1         		            ; 		
        reg		[9:0]                         cnt_row2         		            ; 			
        wire	[2:0]			              row_wr		                    ;
        wire	[2:0]			              row_rd		                    ;
        reg		[1:0]                         cnt_stride_w		                ;
        reg		[1:0]                         cnt_stride_h		                ;
        reg		                		      row0_rd                           ;
        reg		                		      row1_rd                           ;
        reg		                		      row2_rd                           ;
        reg                                   conv_idle                         ;
        reg                                   conv_vld                          ;
        reg                                   last_vld                          ;
        reg		[3:0]        cnt_conv              	        ;
        reg		[4:0]       cnt_kernel              	        ;
[... 426 lines cut ...]
    else if(!full & we)
        wp <= wp+1'b1;
    else
        wp <= wp;
end
// 读操作
always@(posedge clk or negedge rst_n)
begin
    if(!rst_n)
        dout <= {DW{1'b0}};
    else if(!empty & re)
        dout <= ram[rp];
    else
        dout <= dout;
end
//写操作
always@(posedge clk)
begin
    if(!full & we)
        ram[wp] <= din;
    else
        ram[wp] <= ram[wp];
end
endmodule
```

Ravenslofty, 2022-01-15:
Though I don't feel like running a test to check, my suspicion is that because you're not invoking `memory_bram` to turn memories into block RAMs, you're instead turning the whole thing into DFF RAM, and in this case the memory is so big that Yosys runs out of memory trying to map it.

kangliyu1, 2022-01-16:
> Though I don't feel like running a test to check, my suspicion is that because you're not invoking `memory_bram` to turn memories into block RAMs, you're instead turning the whole thing into DFF RAM, and in this case the memory is so big that Yosys runs out of memory trying to map it.

Thank you very much for your reply, hello! When I use the script 1 below to synthesize the same Conv_3.v, it is still "killed" as shown in Figure 1. It is interrupted at the stage of "24.2. Continuing TECHMAP pass." The script below contains the memory_bram you mentioned, but It is still interrupted, and I am very confused about this. No one has replied to me before, so I am sorry for not adding any questions. I hope you can give me some pointers, thank you very much! !

scripts1：
```
# Yosys synthesis script for Conv_3
#########################
# Parse input files
#########################
# Read verilog files
read_verilog -nolatches ./benchmark/Conv_3.v
# Read technology library
read_verilog -lib -specify /home/kly/OpenFPGA/openfpga_flow/tasks/benchmark_sweep/mac_units/a_file/yosys_file/k6_frac_N10_tileable_adder_chain_dpram8K_dsp36_fracff_40nm_cell_sim.v
#########################
# Prepare for synthesis
#########################
# Identify top module from hierarchy
hierarchy -check -top Conv_3
# - Convert process blocks to AST
proc
# Flatten all the gates/primitives
flatten
# Identify tri-state buffers from 'z' signal in AST
# with follow-up optimizations to clean up AST
tribuf -logic
opt_expr
opt_clean
# demote inout ports to input or output port
# with follow-up optimizations to clean up AST
deminout
opt
opt_expr
opt_clean
check
opt
wreduce -keepdc
peepopt
pmuxtree
opt_clean
########################
# Map multipliers
# Inspired from synth_xilinx.cc
#########################
# Avoid merging any registers into DSP, reserve memory port registers first
memory_dff
wreduce t:$mul
techmap -map /home/kly/OpenFPGA/yosys/share/mul2dsp.v -map /home/kly/OpenFPGA/openfpga_flow/openfpga_yosys_techlib/k6_frac_N10_tileable_adder_chain_dpram8K_dsp36_40nm_dsp_map.v -D DSP_A_MAXWIDTH=36 -D DSP_B_MAXWIDTH=36 -D DSP_A_MINWIDTH=2 -D DSP_B_MINWIDTH=2 -D DSP_NAME=mult_36x36
select a:mul2dsp
setattr -unset mul2dsp
opt_expr -fine
wreduce
select -clear
chtype -set $mul t:$__soft_mul# Extract arithmetic functions
#########################
# Run coarse synthesis
#########################
# Run a tech map with default library
techmap
alumacc
share
opt
fsm
# Run a quick follow-up optimization to sweep out unused nets/signals
opt -fast
# Optimize any memory cells by merging share-able ports and collecting all the ports belonging to memorcy cells  
memory -nomap
opt_clean
#########################
# Map logics to BRAMs
[... 174 lines cut ...]
Using template $paramod\_90_alu\A_SIGNED=0\B_SIGNED=0\A_WIDTH=3\B_WIDTH=5\Y_WIDTH=6 for cells of type $alu.
Using template $paramod\_90_alu\A_SIGNED=0\B_SIGNED=0\A_WIDTH=5\B_WIDTH=7\Y_WIDTH=8 for cells of type $alu.
Using extmapper simplemap for cells of type $reduce_and.
Using template $paramod\_90_alu\A_SIGNED=0\B_SIGNED=0\A_WIDTH=4\B_WIDTH=10\Y_WIDTH=10 for cells of type $alu.
Using template $paramod\_90_alu\A_SIGNED=0\B_SIGNED=0\A_WIDTH=5\B_WIDTH=10\Y_WIDTH=10 for cells of type $alu.
Using template $paramod\_90_alu\A_SIGNED=0\B_SIGNED=0\A_WIDTH=2\B_WIDTH=4\Y_WIDTH=4 for cells of type $alu.
Using template $paramod\_90_alu\A_SIGNED=0\B_SIGNED=0\A_WIDTH=7\B_WIDTH=20\Y_WIDTH=20 for cells of type $alu.
Using template $paramod\_90_alu\A_SIGNED=0\B_SIGNED=0\A_WIDTH=1\B_WIDTH=20\Y_WIDTH=20 for cells of type $alu.
Using template $paramod\_90_alu\A_SIGNED=0\B_SIGNED=0\A_WIDTH=1\B_WIDTH=5\Y_WIDTH=5 for cells of type $alu.
Using template $paramod\_90_alu\A_SIGNED=0\B_SIGNED=0\A_WIDTH=4\B_WIDTH=5\Y_WIDTH=5 for cells of type $alu.
Using template $paramod\_90_alu\A_SIGNED=0\B_SIGNED=0\A_WIDTH=1\B_WIDTH=2\Y_WIDTH=2 for cells of type $alu.
Using template $paramod\_90_alu\A_SIGNED=0\B_SIGNED=0\A_WIDTH=1\B_WIDTH=4\Y_WIDTH=4 for cells of type $alu.
Using template $paramod\_90_alu\A_SIGNED=0\B_SIGNED=0\A_WIDTH=4\B_WIDTH=4\Y_WIDTH=4 for cells of type $alu.
Using template $paramod\_90_alu\A_SIGNED=0\B_SIGNED=0\A_WIDTH=1\B_WIDTH=7\Y_WIDTH=7 for cells of type $alu.
Using template $paramod\_90_alu\A_SIGNED=0\B_SIGNED=0\A_WIDTH=6\B_WIDTH=7\Y_WIDTH=7 for cells of type $alu.
Using template $paramod\_90_alu\A_SIGNED=0\B_SIGNED=0\A_WIDTH=7\B_WIDTH=10\Y_WIDTH=10 for cells of type $alu.
Using template $paramod\_90_alu\A_SIGNED=1\B_SIGNED=1\A_WIDTH=20\B_WIDTH=20\Y_WIDTH=21 for cells of type $alu.
Using template $paramod\_90_alu\A_SIGNED=1\B_SIGNED=1\A_WIDTH=20\B_WIDTH=21\Y_WIDTH=22 for cells of type $alu.
Using template $paramod\_90_alu\A_SIGNED=1\B_SIGNED=1\A_WIDTH=20\B_WIDTH=22\Y_WIDTH=23 for cells of type $alu.
Using template $paramod\_90_alu\A_SIGNED=1\B_SIGNED=1\A_WIDTH=20\B_WIDTH=23\Y_WIDTH=24 for cells of type $alu.
Using template $paramod\_90_alu\A_SIGNED=1\B_SIGNED=1\A_WIDTH=20\B_WIDTH=24\Y_WIDTH=24 for cells of type $alu.
Using template $paramod\_90_alu\A_SIGNED=0\B_SIGNED=0\A_WIDTH=11\B_WIDTH=13\Y_WIDTH=32 for cells of type $alu.
Killed

```

Tanvir-Rizve, 2024-02-01:
**Have you got any solution for this problem?**

> Thank you very much for your reply, hello! When I use the script 1 below to synthesize the same Conv_3.v, it is still "killed" as shown in Figure 1. It is interrupted at the stage of "24.2. [...]