YosysHQ/yosys #3152: Citing Yosys

rishubn, 2022-01-08:
The [old Yosys website ](http://bygone.clairexen.net/yosys/documentation.html)states the following:
```
In papers and reports, please refer to Yosys as follows: Clifford Wolf. Yosys Open SYnthesis Suite. http://www.clifford.at/yosys/, e.g. using the following BibTeX code:

@MISC{Yosys,
	author = {Clifford Wolf},
	title = {Yosys Open SYnthesis Suite},
	howpublished = "\url{http://www.clifford.at/yosys/}"
}
```

Since the project has greatly evolved since then, is this still how yosys should be cited?

Ravenslofty, 2022-01-15:
I think @clairexen has indicated a preference to being attributed as "C. Wolf", and the link should point to http://bygone.clairexen.net/yosys/ but I'll defer to her.

clairexen, 2022-01-24:
"C. Wolf" and the website is https://yosyshq.net/yosys.

rishubn, 2022-01-25:
@clairexen Thanks! Will close this issue. I saw this recently, maybe it is worth to add? 

https://docs.github.com/en/repositories/managing-your-repositorys-settings-and-features/customizing-your-repository/about-citation-files

closed 2022-01-25 by rishubn