YosysHQ/yosys #3165: abc buffering doesn't work with techmap cell

antonblanchard, 2022-01-18:
The OpenROAD ASIC flows (Openlane and OpenROAD-flow-scripts) use abc buffering to manage fan out (`buffer -N X`). While looking at some timing issues in https://github.com/The-OpenROAD-Project/OpenLane/issues/870 I noticed that abc buffering seems to ignore `techmap -map YY` cells. I'm not sure if it's a yosys or abc bug, or if we are driving yosys incorrectly.

## Steps to reproduce the issue

The yosys script is:

```
yosys -import
set liberty_file trimmed.lib
read_liberty -lib $liberty_file
read_verilog -sv Microwatt_FP_DFFRFile.v
synth -top Microwatt_FP_DFFRFile
opt -purge

if { [info exists ::env(DO_TECHMAP)] && $::env(DO_TECHMAP) == 1 } {
    techmap -map mux2_map.v
}

dfflibmap -liberty $liberty_file
opt
abc -D 5000 \
    -liberty trimmed.lib \
    -constr regfile.constr \
    -dff \
    -script "+strash;ifraig;scorr;dc2;dretime;strash;&get,-n;&dch,-f;&nf,{D};&put;buffer,-N,5,-v"
setundef -zero
splitnets
opt_clean -purge
hilomap -singleton -hicell sky130_fd_sc_hd__conb_1 HI -locell sky130_fd_sc_hd__conb_1 LO
insbuf -buf sky130_fd_sc_hd__buf_4 A X
write_verilog -noattr -noexpr -nohex -nodec out.v
```

And the `mu2_map.v` is:

```
module \$_MUX_ (
    output Y,
    input A,
    input B,
    input S
    );
  sky130_fd_sc_hd__mux2_1 _TECHMAP_MUX (
      .X(Y),
      .A0(A),
      .A1(B),
      .S(S)
  );
endmodule
```

FYI while debugging this issue, I realised that we don't need to manually techmap the muxes at all and I'll remove that from Openlane. The issue remains for other techmap'ed cells.

If we enable manual techmap'ing of muxes then nets on the mux inputs have massive fan outs:

```
# yosys doit.tcl > /dev/null
# grep '\.' out.v| sort | uniq -c | sort -n|tail -5
      5     .S(_19223_),
      5     .S(_19224_),
      5     .S(_19225_),
      5     .S(_19226_),
   5120     .CLK(CLK),

# DO_TECHMAP=1 yosys doit.tcl > /dev/null
# grep '\.' out.v| sort | uniq -c | sort -n|tail -20
   1280     .S(R3[1]),
   2560     .S(R1[0]),
   2560     .S(R2[0]),
   2560     .S(R3[0]),
   5120     .CLK(CLK),
```

antonblanchard, 2022-01-18:
Complete testcase: [abc-buffer-issue.tar.gz](https://github.com/YosysHQ/yosys/files/7892922/abc-buffer-issue.tar.gz)

Ravenslofty, 2022-01-19:
I suspect the custom cells you're mapping cause Yosys to "hide" (turn into network primary inputs/outputs) the cells from ABC.

What you have here with your mapped cell is a "box" - a cell with logic that one can optionally look inside. ABC9 is box-aware, ABC is not. Yosys can't communicate to ABC "there's this logic in the path you should be aware of", so then ABC cannot distinguish between the inputs and outputs to that cell and, say, a wire going to a GPIO. Since the cell doesn't exist to ABC, it cannot buffer it.

antonblanchard, 2022-01-19:
@Ravenslofty Thanks. I tried a simple test to confirm:

```
module mux2(
    input  i0,
    input  i1,
    input  s0,
    output z
);
    assign z = (s0) ? i1 : i0;
endmodule
```

With a `mux2_map.v`:

```
module \$_MUX_ (
    output Y,
    input A,
    input B,
    input S
    );
  sky130_fd_sc_hd__mux2_1 _TECHMAP_MUX (
      .X(Y),
      .A0(A),
      .A1(B),
      .S(S)
  );
endmodule
```

```
yosys -import
set liberty_file trimmed.lib
read_liberty -lib $liberty_file
read_verilog -sv mux2.v
synth -top mux2
opt -purge

if { [info exists ::env(DO_TECHMAP)] && $::env(DO_TECHMAP) == 1 } {
    techmap -map mux2_map.v
}

dfflibmap -liberty $liberty_file
opt
abc -liberty trimmed.lib \
    -script "+strash;ifraig;scorr;dc2;dretime;strash;&get,-n;&dch,-f;&nf,{D};&put;buffer,-N,5,-v"
```

liberty file is from the tarball above. Looking at the `input.blif` we feed to abc, with the mux2 techmap disabled I see 

```
.model netlist
.inputs ys__n0 ys__n1 ys__n2
.outputs ys__n3
# ys__n0     \i0
# ys__n1     \i1
# ys__n2     \s0
# ys__n3     \z
.names ys__n0 ys__n1 ys__n2 ys__n3
1-0 1
-11 1
.end
```

And with it disabled, it is completely missing:

```
.model netlist
.inputs dummy_input

.outputs
.end
```

antonblanchard, 2022-01-19:
Is there anything stopping us from using abc9? It looks it needs an option to pass liberty files to abc, but not sure if there is more work than that.

antonblanchard, 2022-01-19:
Answering my own question:

```
ERROR: ABC9 flow only supported for FPGA synthesis (using '-lut' option)
```

antonblanchard, 2022-01-24:
We've removed the manual mux mapping from Openlane, but this is still going to be an issue for things we have to tech map (eg full/half adders). I presume it is also an issue with modules that explicitly use standard cells. As such it would be good to fix.

@eddiehung Is there anything preventing us getting abc9 to work without LUTs? I have no idea if it is an enormous amount of work or not.

eddiehung, 2022-01-24:
I'll be honest and say I don't know much about the ASIC side of ABC. What I can see is that you are using some abc9 functions in your script (specifically `&nf`) but you are entering from regular abc (non box aware flow) via BLIF. Having no familiarity with `&nf` I can't tell you if it is box aware (many `&`-prefix commands are not necessarily box aware) -- this would be a question for upstream I'm afraid.