YosysHQ/yosys #3169: Equivalence checking of Verilog module fails

Simon-Lever, 2022-01-27:
I am trying to verify changes I made to a Verilog module by using the yosys equivalence checking.
As a first step I am trying to compare the unmodified module with itself in order to make sure the equivalence checking is working out.

The module I am currently checking is found here:

[dma_ram_demux_wr.v](https://github.com/alexforencich/verilog-pcie/blob/master/rtl/dma_ram_demux_wr.v)

The commands I am using in order to get the checking started are:

```
read_verilog dma_ram_demux_wr.v
prep -flatten -top dma_ram_demux_wr
splitnets -ports;;
design -stash gold
read_verilog dma_ram_demux_wr.v
prep -flatten -top dma_ram_demux_wr
splitnets -ports;;
design -stash gate
design -copy-from gold -as gold dma_ram_demux_wr
design -copy-from gate -as gate dma_ram_demux_wr
equiv_make gold gate equiv
prep -flatten -top equiv

opt_clean -purge
show -prefix equiv-prep -colors 1 -stretch

opt -full
equiv_simple -seq 5
equiv_induct -seq 5
equiv_status -assert
```

Eventually the check fails with the following message:

> ERROR: Found 196 unproven $equiv cells in 'equiv_status -assert'.

I have tried increasing the number of steps involved up to 200, which didn't improve the situation.
For every step I am getting the following messages:

> Proving existence of base case for step 1. (35677 clauses over 14018 variables)                                                                                                                                                               > Proving induction step 1. (79153 clauses over 30375 variables)                                                                                         
> Proof for induction step failed. Extending to next time step.

It seems like Yosys is not able to solve the clauses. Am I missing something important here?

Ravenslofty, 2022-01-29:
```
read_verilog dma_ram_demux_wr.v
prep -flatten -top dma_ram_demux_wr
splitnets -ports;;
design -stash gold
read_verilog dma_ram_demux_wr.v
prep -flatten -top dma_ram_demux_wr
splitnets -ports;;
design -stash gate
design -copy-from gold -as gold dma_ram_demux_wr
design -copy-from gate -as gate dma_ram_demux_wr
equiv_make gold gate equiv

equiv_simple -seq 5
equiv_induct -seq 5
equiv_status -assert
```

```
9. Executing EQUIV_INDUCT pass.
Found 1114 unproven $equiv cells in module equiv:
Warning: No SAT model available for cell genblk1[1].fifo_sel ($mem_v2).
Warning: No SAT model available for cell genblk1[0].fifo_sel ($mem_v2).
  Proving existence of base case for step 1. (51125 clauses over 19974 variables)
  Proving induction step 1. (110049 clauses over 42287 variables)
  Proof for induction step holds. Entire workset of 1114 cells proven!
No selected unproven $equiv cells found in gate.
No selected unproven $equiv cells found in gold.
Proved 1114 previously unproven $equiv cells.

10. Executing EQUIV_STATUS pass.
Found 1522 $equiv cells in equiv:
  Of those cells 1522 are proven and 0 are unproven.
  Equivalence successfully proven!
```

Simon-Lever, 2022-01-31:
Great, yes that was the issue, thank you!

I have now tried doing the same with my modified version of the module and it fails to prove some signals.
Is there any way I can find out under which condition the (e.g. state of the input pins) the particular signals differ? How can I get an idea of what is going wrong?

PR #3502 (2022-10-11): equiv_opt and clk2fflogic fixes
Fix and re-enable essentially all equiv_opt based tests. This is based on #2740 but instead of changing the negative hold time behavior of clk2fflogic it fixes the underlying issue in the logic used to emulate negative hold time using $ff cells.

Apart from fixing the previously disabled tests in the test suite, this also fixes the following issues:

* #639 remains fixed, even after now having reverted the incorrect previous fix for this
* fixes #2728
* fixes #3169 
* fixes #3428
closed 2022-10-11 by jix