YosysHQ/yosys #3190: yosys frontend cannot handle delays with a specified unit

ted-xie, 2022-02-08:
## Steps to reproduce the issue

After unzipping the attached zip archive:
`yosys -p 'read_verilog case.v'` -> succeeds.
`yosys -p 'read_verilog case_bad.v'` -> fails.

My yosys version:
```
$ yosys -version
Yosys 0.9+4052 (git sha1 0ccc7229, clang 11.0.1-2 -fPIC -Os)
```

## Expected behavior

I would expect that at least in the read_verilog stage, `q <= #1 d` would be treated the same way as `q <= #1ns d`. `read_verilog` should at least pass if I specify a unit on the delay in a nonblocking assignment.

## Actual behavior

`read_verilog` fails because it expects another token. See error message below.

```
Parsing Verilog input from `case_bad.v' to AST representation.
case_bad.v:8: ERROR: syntax error, unexpected TOK_ID
```

ted-xie, 2022-02-08:
[yosys_delay_case.zip](https://github.com/YosysHQ/yosys/files/8028319/yosys_delay_case.zip)

I somehow messed up attaching the test case. See above attachment.

udif, 2022-02-10:
First, I think #1 is not 1 clock cycle but 1 in the current timescale.
Second, this notation is very common, not to add real delay to the design but to make it easier to look at your waveforms and know for sure what is the signal value at clock rise time. so ignoring this or failing *does* make a difference.
In the codebase I work on my day job, we simply define:

`define D #1 

For simulation, and

`define D

For synthesis.

zachjs, 2022-02-11:
I will take a pass at adding support for time literals in delays. For simplicity, the implementation will likely be a bit more permissive than the official grammar by allowing number literals in general rather than just unsigned and fixed-point numbers.

PR #3198 (2022-02-14): verilog: support for time scale delay values
Fixes #3190. @ted-xie
closed 2022-02-14 by zachjs