YosysHQ/yosys #3195: Rename flip-flops with custom suffix

bfg86, 2022-02-10:
I'd like to have meaningful names for flip-flops (makes it easier for debugging). So I tried 
`rename -wire t:$*DFF*`
before dfflibmap. 
This gave names with the flop type as a suffix, for example
`\ready$_DFFE_PP1P_`
I made a patch so its possible to have a custom suffix when running rename -wire. Wondering if anyone else are interested in this patch.
This would make it easier to debug a netlist, and also do name-matching vs RTL in third-party tools (like LEC).

bfg86, 2022-02-10:
After applying the patch you can do
`rename -wire -suffix _reg t:$*DFF*`
and the same register would be called 
`ready_reg`
in the netlist.

dineshannayya, 2022-05-25:
Is this option give user have cell/FF in netlist with same RTL ?
My issue Yosys is:  Yosys netlist generation does not follow the industry standard naming conversion in
preserve cell instance name
Example: Currently in netlist cells instance name with number like _06637_
sky130_fd_sc_hd__nand2_2 _06637_ ( .A(_01599_), .B(_01620_), .Y(_01621_));

Preferred format <rtl_cell_name>_reg : This helps in quick gate-level, ECO and LEC and
STA debug.

Is there is example scripts to show which place this  command need to applied .

dineshannayya, 2022-05-25:
Is is custom script integrated into yosys main branch ? 

 I see tool gives with below error while integrating this patch
ERROR: TCL interpreter returned an error: wrong # args: should be "rename oldName newName"

nakengelhardt, 2022-05-27:
We don't want to provide anything that looks like a standard because we don't have any way to guarantee consistent naming of these cells in situations more complex than a simple assignment to a FF (e.g., if there are any bit selections, slices, concatenations). There will always be cases where the names would not match what another tool will do, or even what a different combination of yosys passes will create. Therefore no one should build any script relying on those names, and if they are only read by a human then it should not make a difference if it says <wirename>_reg or <wirename>$dff.

dineshannayya, 2022-06-04:
Current yosys cell naming scheme (instance name with number) is a critical drawback compare to commercial graded tools.
Netlist with cell instance with matching with source RTL name is critical function to debug the LEC failure, STA analysis and Gate level simulation and debug.

bfg86, 2022-06-05:
@nakengelhardt It does make a difference, especially if you're using Verilog as backend. The '$' character is illegal in Verilog, so all names with '$' are escaped. This makes it hard to read, and as you point out hard to write scripts for.
The patch doesn't change any default behavior, but provides an option to add whatever prefix you'd like.

dineshannayya, 2022-06-05:
@bfg86 I am looking for way to keep cell instance name matching as in RTL. Not sure how rename rule helps here when user will no access to RTL name for a give cell instance.

bfg86, 2022-06-13:
@dineshannayya The rename -wire renames the selection based on the name of the wire that the cell is driving. 

Guessing a bit here: do you have a tech cell instantiated in your RTL, and are you looking for something like dont_touch or size_only in Design Compiler? (guessing since you mentioned a NAND gate in your example)

jix, 2022-06-13:
> ERROR: TCL interpreter returned an error: wrong # args: should be "rename oldName newName"

Note that this message comes from the tcl builtin `rename` not the yosys `rename` command. Errors from yosys commands will not be prefixed by `TCL interpreter returned an error`. How to invoke the yosys `rename` command from tcl is mentioned in `yosys -h tcl`:

> Yosys commands 'proc' and 'rename' are wrapped to tcl commands 'procs' and 'renames' in order to avoid a name collision with the built in commands.

closed 2022-06-20 by nakengelhardt

prajnaputhran17, 2023-07-08:
I am facing the same issue of flipflops renaming after synthesis. Could you tell how to retain(get) names of the filpflops after synthesis same as RTL.

sky130_fd_sc_hd_dfrtp_2 _14258 (
    .CLK(CLK),
    .D(00157),
    .Q(mv_output[20]),
    .RESET_B(RST_N)
  );

Vinayakamk, 2023-07-15:
> Reference

@dineshannayya respected sir,
we faced the same issue of what u faced about registers naming,could u tell us how did u fixed.