YosysHQ/yosys #3205: dual port blockram fails to synthesize ECP5

rowanG077, 2022-02-16:
## Steps to reproduce the issue

I have a true dual port block ram module as generated by the clash-compiler:

```
/* AUTOMATICALLY GENERATED VERILOG-2001 SOURCE CODE.
** GENERATED BY CLASH 1.6.1. DO NOT MODIFY.
*/
`timescale 100fs/100fs
module TopEntity_topEntity_trueDualPortBlockRamWrapper
    ( // Inputs
      input  clkA // clock
    , input  enA
    , input  weA
    , input [15:0] addrA
    , input [23:0] datA
    , input  clkB // clock
    , input  enB
    , input  weB
    , input [15:0] addrB
    , input [23:0] datB

      // Outputs
    , output wire [47:0] result
    );

  // trueDualPortBlockRam begin
  // Shared memory
  reg [24-1:0] mem [65536-1:0];

  reg [23:0] data_slow;
  reg [23:0] data_fast;

  // Port A
  always @(posedge clkA) begin
      if(enA) begin
          data_slow <= mem[addrA];
          if(weA) begin
              data_slow <= datA;
              mem[addrA] <= datA;
          end
      end
  end

  // Port B
  always @(posedge clkB) begin
      if(enB) begin
          data_fast <= mem[addrB];
          if(weB) begin
              data_fast <= datB;
              mem[addrB] <= datB;
          end
      end
  end

  assign result = {data_slow, data_fast};

  // end trueDualPortBlockRam
endmodule
```

Running yosys works without issues:
```
yosys -p "synth_ecp5 -top TopEntity_topEntity_trueDualPortBlockRamWrapper -json ./synth.json" ./TopEntity_topEntity_trueDualPortBlockRamWrapper.v
```

But when running nextpnr with the command:

```
nextpnr-ecp5 --json ./synth.json --25k --package CABGA256 --lpf-allow-unconstrained

```

It fails with the message:

```
ERROR: cell type '$mem_v2' is unsupported (instantiated as 'mem')
```

I assume this means ECP5 doesn't support a true dual port blockram right now? Is there anything I can do to help make it supported :)?

modwizcode, 2022-04-05:
The current BRAM mapper in Yosys does not support mapping to true dual-port BRAMs. This is part of the improvements that @mwkmwkmwk is working to bring with an improved pass at some point in the future.

mwkmwkmwk, 2022-04-05:
oh, I forgot about this issue, actually

this should be already working on my #3189 branch, would you like to be a test subject?

rowanG077, 2022-05-04:
@mwkmwkmwk I have tried to use your branch but `ecppack` fails with the message:

```
terminate called after throwing an instance of 'std::runtime_error'
  what():  no word named 'IOLOGICA.DELAY.DEL_VALUE'
```

See this [gist](https://gist.github.com/rowanG077/e48c80ccb28633d36d71ca4d27a09d6b) for the output of yosys, nextpnr and ecppack one after the other. For reference I used these versions:

- trellis: https://github.com/YosysHQ/prjtrellis/commit/64b38dfda6f0bc556839b513820f943f64b23cc3
- yosys: https://github.com/YosysHQ/yosys/commit/39c12212cabdd8550d4dc3ce8e22539522cb8e8e
- nextpnr: https://github.com/YosysHQ/nextpnr/commit/a4949826465fdc5fde89a0f52cc4078fae319cf3

Essentially the most recent versions of everything at the time of writing.

gatecat, 2022-05-04:
That error doesn't seem at all related to the BRAM inference patch, I'll have a look.

mwkmwkmwk, 2022-05-04:
The pass still prints the final mapping rule picked, and using `yosys -g` dumps a bit more detail about the alternatives considered. Is there some other information you'd want to see in the output?

As for output registers, this pass supports a single register for synchronous ports just like the old one, only now with a few more features on it.  Second output register (aka pipelining register) is still not supported (and is out of scope for `memory_libmap` specifically — it's been decided long ago to do it as a follow-up pass executed later in the flow)

PR #3189 (2022-05-18): Add memory_libmap pass.
Fixes #1959.
Fixes #1134.
Fixes #1958.
Fixes #1836.
Fixes #1101.
Fixes #1087.
Fixes #1088.
Fixes #1802.
Fixes #1550.
Fixes #3205.
closed 2022-05-18 by mwkmwkmwk