YosysHQ/yosys #3217: Yosys crashes when an error encountered under Windows/Anaconda during file delete

LispEngineer, 2022-02-25:
Thank you kindly for Yosys and sorry for the trouble!

## Steps to reproduce the issue

`yowasp-yosys -p "read_verilog -sv crash_yosys.sv; proc; opt; write_json teroshdl_yosys_output.json; stat"`

`crash_yosys.sv`:
```verilog
`timescale 1 ns / 100 ps
module crash_yosys();

// Simulator generated clock & reset
logic  clk;
initial begin
  clk <= 1'b0;
end

always begin
  #10 clk <= ~clk;
end

// Only run the I2C transaction once
always_ff @(posedge clk) begin
  $stop;
end

endmodule
```

Not sure how to do this with the current git master (not much of a Python user, sorry). Using:
` Yosys 0.13+39 (git sha1 9c9366895, ccache clang 11.0.0-2~ubuntu20.04.1 -Os -flto -flto)`
Pip3 reports: `yowasp-yosys  0.13.post39.dev297`

and also
``

Windows 10, using Anaconda 3, and a virtual environment with Python 3.9.7 and pip-installed Yosys. Also tried with a Python 3.8.12 virtual environment.

## Expected behavior

Normal termination without a crash deep in Python.

## Actual behavior

Output:
```
1. Executing Verilog-2005 frontend: crash_yosys.sv
crash_yosys.sv:0: ERROR: System task `$stop' outside initial block is unsupported.
Traceback (most recent call last):
  File "C:\bin\anaconda3\envs\teroshdl\lib\shutil.py", line 628, in _rmtree_unsafe
    os.rmdir(path)
PermissionError: [WinError 32] The process cannot access the file because it is being used by another process: 'C:\\Users\\Doug\\AppData\\Local\\Temp\\tmpc4boy9k2yosys'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\bin\anaconda3\envs\teroshdl\lib\tempfile.py", line 801, in onerror
    _os.unlink(path)
PermissionError: [WinError 5] Access is denied: 'C:\\Users\\Doug\\AppData\\Local\\Temp\\tmpc4boy9k2yosys'

During handling of the above exception, another exception occurred:
```

Then the above two things repeat hundreds of times until it finally quits with `RecursionError: maximum recursion depth exceeded while calling a Python object` and a long backtrace.

Same thing happens with `yowasp-yosys --version` as well.

whitequark, 2023-02-04:
Fixed as https://github.com/YoWASP/yosys/issues/27.

whitequark, 2023-02-04:
Note that the binaries with the fix aren't uploaded yet; they'll be in place in a few days.

closed 2023-02-04 by whitequark