YosysHQ/yosys #3218: Difference in results depending on whether Yosys is compiled with Clang or GCC?

tcal-x, 2022-02-26:
## Steps to reproduce the issue

Unpack the attached zip.

Run the command in `run` first using a Yosys compiled with clang, then with a Yosys compiled with gcc.   The results will be different.  

## Expected behavior

Perhaps I'm being naive, but I would expect the output to be the same regardless of how Yosys was compiled (assuming there were no differences in -D defines)

## Actual behavior

There were differences that seemed significant in the overall design; when using Yosys compiled with Clang, the max frequency seemed better on average (after nextpnr).

From the reduced design that's included in the zip, enough to show a difference:

With Clang:
```
   Number of wires:               3817
```
With GCC:
```
   Number of wires:               3815
```

tcal-x, 2022-02-26:
[repro.zip](https://github.com/YosysHQ/yosys/files/8145879/repro.zip)

nakengelhardt, 2022-06-20:
Your script calls ABC, which is well known to be highly sensitive to just about everything in the design. Different compilers lead to different iteration orders in some places (e.g. when sets are used) which can lead to IDs being assigned differently.

In fact to account for that effect when benchmarking we're adding a way to shuffle the private names in #3277 - that's enough to get completely different results.