YosysHQ/yosys #3225: Synthesizing large memory blocks takes a really long time.

hughperkins, 2022-03-04:
## Steps to reproduce the issue

e.g. synthesize:

```
// represents memory controller (plus memory that the memory is controlling)
module mem(
    input we, clk,
    input [15:0] read_addr,
    input [15:0] write_addr,
    output [15:0] read_data,
    input [15:0] write_data,

    input [15:0] oob_write_addr,
    input [15:0] oob_write_data,
    input oob_wen
);
    reg [15:0] mem [65536];

    always @(posedge clk) begin
        if (we) begin
            mem[write_addr] <= write_data;
        end
        if(oob_wen) begin
            mem[oob_write_addr] <= oob_write_data;
        end
    end
    assign read_data = mem[read_addr];
endmodule
```

And use following yosys script:
```
read_verilog -sv src/mem.sv
synth
```

## Expected behavior

Given that it's a single memory block, I would have expected that one could synthesize for one memory location, and just replicate that 65,526 times for the other memory locations?

## Actual behavior

Treats each memory location as its own special snowflake, running full optimization passes on every single memroy location, which takes a really long time.

Ravenslofty, 2022-03-04:
Well, `synth` calls `memory_map` which builds a memory block out of DFFs and MUXes, and then it's done. Then `opt` gets called and runs on every one of those DFFs and MUXes. So, this is a case where the philosophy of "do one thing and do it well" within the Yosys subcommands doesn't work.

However I'd argue it's not necessarily a realistic example: FPGAs and ASICs would map these to hard memory blocks using `memory_bram` anyway.

hughperkins, 2022-03-04:
what is `memory_bram`? If I google, I just find links talking about Xilinx fpgas, and also a link into yosys sourcecode, https://github.com/YosysHQ/yosys/blob/master/passes/memory/memory_bram.cc . I searched through https://www.eg.bucknell.edu/~csci320/2016-fall/wp-content/uploads/2015/08/verilog-std-1364-2005.pdf for `bram`, but found nothing. I'm targeting ASIC, not fpga, and want to use standard verilog, nothing specific to particular FPGAs, or to yosys. Is `memory_bram` still appropriate for such usage?

Ravenslofty, 2022-03-04:
`memory_bram` is a Yosys pass; you can read about the format it uses with `yosys -p "help memory_bram"`.

You're going to need to be more specific with "standard Verilog"; if you mean that you're writing standard Verilog as input then I think it's reasonable to use `memory_bram`.

When targeting ASICs, I think one would use a memory compiler to build it out of SRAM cells etc, producing a hard macro. One then writes a rules file to tell Yosys about this hard macro, and then uses `memory_bram` to map the memory to that hard macro.

In that respect, there's nothing different between FPGAs and ASICs.

hughperkins, 2022-03-05:
Interesting. Good information. Thanks! (I'm just beginning :) ).

closed 2022-03-05 by hughperkins