YosysHQ/yosys #3235: unexpected behevior of  result occurs

1353369570, 2022-03-17:
## Steps to reproduce the issue

Consider the following piece of code:

```
module top  (y, clk, wire3, wire2);
  output wire [(32'h50):(32'h0)] y;
  input wire [(1'h0):(1'h0)] clk;
  input wire signed [(5'h15):(1'h0)] wire3;
  input wire signed [(2'h3):(1'h0)] wire2;
  reg signed [(5'h15):(1'h0)] reg10 = (1'h0);
  assign y = {reg10};
  always
    @(posedge clk) reg10 <= (-$signed((-(wire2 << {wire3}))));
endmodule
```

Run with yosys version

       Generated by Yosys 0.9+4081 (git sha1 UNKNOWN, clang 10.0.0-4ubuntu1 -fPIC -Os)

and the following command

     yosys -p 'read_verilog  rtl.v; synth; write_verilog  syn_yosys.v'

## Expected behavior

    when Executing CHECK pass and Verilog backend,get a corresponding Write Design to Verilog file

## Actual behavior

    The corresponding Verilog file is an empty function
 `/* Generated by Yosys 0.9+4081 (git sha1 UNKNOWN, clang 10.0.0-4ubuntu1 -fPIC -Os) */

module \$abstract\top ();
endmodule`

Ravenslofty, 2022-03-18:
Yosys 0.9+4081 is nine months old or so, could you retry with latest Yosys?

1353369570, 2022-03-18:
这是来自QQ邮箱的假期自动回复邮件。收到。                  祝好！                  张漪

1353369570, 2022-03-23:
> Yosys 0.9+4081 已经九个月左右了，你能用最新的 Yosys 重试吗？

The latest yosys has been tried, and the result is expected, but the above situation occurs in the 0.9 version, does it indicate that there is a bug in the yosys 0.9+4081

Ravenslofty, 2022-03-23:
More generally you probably need to run `synth -top top` rather than `synth`.

1353369570, 2022-03-23:
> synth -top top
Yes, executing "synth-top top" on Yosys 0.9+4081 can get   desired results, but can you tell me what is causing the problem when executing the command 'synth' directly?

mmicko, 2022-04-08:
@1353369570 In exactly that time there were multiple PRs to refactor and  improve hierarchy pass, so you probably have some non-stable build in middle of these change. Closing issue since this is not present any more.

closed 2022-04-08 by mmicko