YosysHQ/yosys #3239: cmake support

nathanprat, 2022-03-21:
Is there any traction for CMake support?
I know about https://github.com/YosysHQ/yosys/issues/552 (and https://github.com/YosysHQ/yosys/pull/3237) but it is 2022 and I thought I would ask again.

Current pain points with the build system:
- no way to build a proper STATIC lib (= .a) with the current Makefile
    - "config-gcc-static" FAIL, and in any case it is not a static build
- the generated libyosys.so has a wrong SONAME(= $(LIBDIR)/libyosys.so) when it SHOULD be libyosys.so
 which makes it hard to properly package libyosys (without resorting to patchelf)
- no way to properly package/distribute as all paths are essentially hardcoded
- a Makefile is NOT made to handle dependencies; that is what autoconf is for
- configuring the Makefile requires generating a file on disk (Makefile.conf) which is less than ideal when
integrating with CMake downstream projects

As it is pretty stupid to maintain two separate build systems, I would like to know if the maintainers are interested in migrating to CMake?
- if yes: I would gladly make a PR; in which case I would simply refactor all Makefile/Makefile.inc in the project
- if no: I will simply maintain a "quick and dirty" fork with CMake; It works fine but it is only for LIBYOSYS, it does run the tests, it does not handle the various options, etc

jonnysoe, 2023-02-07:
@nathanprat,
Nice work on the initial CMake!

I just wanted to add on some of my observations on the difficult to package issue...

I found it the hard way as I move the library file around, but the linker just adamantly change the ELF's dependent libyosys.so SONAME regardless, I adjusted the `PREFIX` and found that it changes, until I realize something is wrong with the ELF as the linker will just take that SONAME.

As you've mentioned, tt seems that SONAME was hardcoded to `LIBDIR` which changes as we use different `PREFIX`, and by default it is always there - `/usr/local`. I should have stumbled on your pages earlier :)

Not sure if there are discussions or issues on this yet, perhaps this should be addressed as well?

nathanprat, 2023-03-01:
Hi,

I did not, not really. For a while I was using `-whole-archive` cf https://github.com/YosysHQ/yosys/issues/3241 and it worked fine but I end up packaging libyosys as a shared lib instead (with CMake CPack).

A proper fix would require touching the code and that's not something I want to do.

jonnysoe, 2023-03-03:
@nathanprat ,
I've just read that the better option is to [compile as object library](https://stackoverflow.com/a/72368810/19336104) and link as objects instead of static library, it is less clunky than `enable-whole-archive-link-disable`.
I've tried various methods including modifying code, this seems more reliable with the exception that we do not chain-linking libraries, eg.
```
exe_A -> lib_A -> lib_B -> libyosys
```

ctormuthu, 2023-05-19:
+1 to this request. 

Changing flags etc. in Makefile will require you to run make clean and then run make; this is problematic and needs careful handling on all platforms; moving to CMake is going to help Yosys be more modern and multiplatform with ease

hcook95, 2025-04-17:
Apologies for reviving this old issue, but @n-prat, where could I find your initial efforts on CMake?

n-prat, 2025-04-17:
Hi @hcook95 

You can find the work in this repo https://github.com/Interstellar-Network/yosys/branches/all?query=cmake&lastTab=overview but this is really **not** "production grade".
Also, as you can see, this is not at all maintained, so I am not even sure it compiles with recent gcc/clang or CMake.

KrystalDelusion, 2025-04-17:
@hcook95 there was some recent effort internally on a cmake conversion that you can find here: https://github.com/YosysHQ/yosys/tree/micko/cmake

It's still missing a few things before it can fully replace the current Makefile approach, but the intention is to continue the work later.

whitequark, 2025-04-17:
From my perspective, a CMake buildsystem would be very much appreciated. Among other things it would enable development on Windows without MinGW or Cygwin.

nickrallison, 2025-04-19:
I would say I would also really appreciate that, I'm working on Rust bindings for yosys and having the ability to natively output a libyosys.dll, or .dylib file would be really handy.

KrystalDelusion, 2025-10-07:
I think #5204 has the most up-to-date code/discussion

whitequark, 2026-03-19:
There is no public dev branch for CMake. I will push my changes once I'm done with the UK settlement process in a few weeks.