YosysHQ/yosys #3242: The alumacc pass does not preserve attributes

tdene, 2022-03-21:
### Description

The alumacc pass maps `+` to the `$alu` cell.

Attributes may be assigned to the `+` operator in the source HDL.

Any such attributes are not passed on to the `$alu` cell. This is understandable to a degree,
as the `$alu` cell can map multiple operators to the same cell, and one would have to handle
possible attribute collision.

However, there currently is no plausible way to retrieve the attributes in question and associate
the original `$add` with its corresponding `$alu`
Even without an actual attribute map, the code should preserve this information in some
retrievable fashion.

## Steps to reproduce the issue

Take this sample source code:
```
module demo(cout, sum, a, b, cin);
  
    input  [15:0] a, b;
    input cin;
    output cout;
    output [15:0] sum;

    assign {cout,sum} = a + (* test_attribute *) b + cin;

endmodule
```

Run it through a quick script:
```
read -sv demo.v
hierarchy -check -top demo
select -module demo
alumacc
```

## Actual behavior

```
/* Generated by Yosys 0.15+11 (git sha1 cbece4af0, gcc 9.4.0-1ubuntu1~20.04 -fPIC -Os) */

(* cells_not_processed =  1  *)
(* src = "demo.v:1.1-10.10" *)
module demo(cout, sum, a, b, cin);
  (* src = "demo.v:8.25-8.51" *)
  wire [16:0] _0_;
  (* src = "demo.v:8.25-8.57" *)
  wire [16:0] _1_;
  wire [16:0] _2_; 
  wire [16:0] _3_;
  (* src = "demo.v:3.19-3.20" *)
  input [15:0] a;
  wire [15:0] a;
  (* src = "demo.v:3.22-3.23" *)
  input [15:0] b;
  wire [15:0] b;
  (* src = "demo.v:4.11-4.14" *)
  input cin;
  wire cin;
  (* src = "demo.v:5.12-5.16" *)
  output cout;
  wire cout; 
  (* src = "demo.v:6.19-6.22" *)
  output [15:0] sum;
  wire [15:0] sum;
  assign _0_ = a + (* src = "demo.v:8.25-8.51" *) (* test_attribute = 32'd1 *) b;
  (* src = "demo.v:8.25-8.57" *)
  \$alu  #(
    .A_SIGNED(32'd0),
    .A_WIDTH(32'd16),
    .B_SIGNED(32'd0),
    .B_WIDTH(32'd16),
    .Y_WIDTH(32'd17)
  ) _5_ (
    .A(b),
    .B(a),
    .BI(1'h0),
    .CI(cin),
    .CO(_3_), 
    .X(_2_),
    .Y(_1_)
  );
  assign { cout, sum } = _1_; 
endmodule
```

## Expected behavior

One or more of these behaviors is expected:
```
  assign _0_ = a + (* src = "demo.v:8.25-8.51" *) (* test_attribute = 32'd1 *) b;
  (* src = "demo.v:8.25-8.57" *) (* test_attribute = 32'd1 *)
  \$alu  #(
```

```
  assign _0_ = a + (* src = "demo.v:8.25-8.51" *) (* test_attribute = 32'd1 *) (* corresponds_to_alu = 1 *) b;
  (* src = "demo.v:8.25-8.57" *) (* corresponds_to_alu = 1 *)
  \$alu  #(
```

tdene, 2022-03-21:
While the `src` attribute may be used as a crude correspondence map in this case, this is not good coding practice and may easily lead to edge-cases that do not behave as expected.

Ravenslofty, 2022-03-24:
What should happen in the case of two attributes having conflicting values? e,g. 
```verilog
module demo(cout, sum, a, b, cin);
  
    input  [15:0] a, b;
    input cin;
    output cout;
    output [15:0] sum;

    assign {cout,sum} = a + (* test_attribute = "foo" *) b + (* test_attribute = "bar" *) cin;

endmodule
```

Sure, it's not a problem for your case, as I understand it, but it's a problem in the general case.

tdene, 2022-03-24:
Agreed, this was my main concern above:
> as the $alu cell can map multiple operators to the same cell, and one would have to handle
> possible attribute collision.

I don't think collision is ever a problem that can be solved elegantly, hence my suggestion of just:
```
  assign _0_ = a + (* src = "demo.v:8.25-8.51" *) (* test_attribute = 32'd1 *) (* alu_src = 1 *) b;
  (* src = "demo.v:8.25-8.57" *) (* alu_src = 1 *)
  \$alu  #(
```
Because after all, the attribute doesn't necessarily need to be transferred to the `$alu` block. `alumacc` leaves behind the original `+` statement until some subsequent pass `clean`s it out. As long as there is a way to associate the two, the original `$add` with the transformed `$alu`, code can be written to traverse the bread-crumb trail. The problem is just that currently there is no bread-crumb trail at all.

Now, introducing an extra attribute is far from the most beautiful code.

By raising these issue, I was mainly hoping for someone to tell me "there's this hidden way you can figure out which `$alu` belongs to which `$add`.

As mentioned [here](https://github.com/YosysHQ/yosys/issues/3242#issuecomment-1074244273), the `src` attribute can be used as a crude bread-crumb trail, by just assuming "if an `$alu` and an `$add` were `src`d from the same file and line number, they are likely associated". I feel like that's a horrible assumption to make though.