YosysHQ/yosys #3244: synth_intel_alm: Crash on design with inferred memory

maikmerten, 2022-03-24:
## Steps to reproduce the issue

Unpack ```bram_wb8.v``` from the attacked ZIP archive. Start yosys and execute following commands:

```
   read_verilog bram_wb8.v
   synth_intel_alm
```

## Expected behavior

The ```synth_intel_alm``` command should finish without yosys crashing.

## Actual behavior

Yosys crashes during ```synth_intel_alm``` with a thrown ```std::out_of_range```:
```
2.50.13. Executing AIGMAP pass (map logic to AIG).
Module bram_wb8: replaced 0 cells with 0 new cells, skipped 47 cells.
  not replaced 7 cell types:
       2 $_NOT_
       2 $_AND_
       8 MISTRAL_M10K
      24 MISTRAL_IB
       9 MISTRAL_OB
       1 MISTRAL_CLKBUF
       1 $__MISTRAL_FF_SYNCONLY

2.50.13.1. Executing ABC9_OPS pass (helper functions for ABC9).

2.50.13.2. Executing ABC9_OPS pass (helper functions for ABC9).

2.50.13.3. Executing XAIGER backend.
terminate called after throwing an instance of 'std::out_of_range'
  what():  vector::_M_range_check: __n (which is 9) >= this->size() (which is 1)
```

The backtrace:

```
(gdb) bt
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
#1  0x00007ffff783b859 in __GI_abort () at abort.c:79
#2  0x00007ffff7e59911 in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#3  0x00007ffff7e6538c in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007ffff7e653f7 in std::terminate() () from /lib/x86_64-linux-gnu/libstdc++.so.6
#5  0x00007ffff7e656a9 in __cxa_throw () from /lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x00007ffff7e5c3ab in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#7  0x0000000000e73a1d in std::vector<Yosys::RTLIL::SigBit, std::allocator<Yosys::RTLIL::SigBit> >::_M_range_check (this=0x144bb70, __n=9)
    at /usr/bin/../lib/gcc/x86_64-linux-gnu/9/../../../../include/c++/9/bits/stl_vector.h:1070
#8  std::vector<Yosys::RTLIL::SigBit, std::allocator<Yosys::RTLIL::SigBit> >::at (this=0x144bb70, __n=9) at /usr/bin/../lib/gcc/x86_64-linux-gnu/9/../../../../include/c++/9/bits/stl_vector.h:1109
#9  Yosys::RTLIL::SigSpec::operator[] (this=0x144bb48, index=<optimized out>) at ./kernel/rtlil.h:876
#10 (anonymous namespace)::XAigerWriter::XAigerWriter (this=0x7fffffffa1b8, module=0x1587070, dff_mode=false) at backends/aiger/xaiger.cc:284
#11 0x0000000000e6f49d in (anonymous namespace)::XAigerBackend::execute (this=<optimized out>, f=@0x7fffffffa6f0: 0x161a780, filename=..., args=..., design=0x1341120) at backends/aiger/xaiger.cc:796
#12 0x00000000005a006c in Yosys::Backend::execute (this=0x12d2298 <(anonymous namespace)::XAigerBackend>, args=..., design=0x1341120) at kernel/register.cc:644
#13 0x000000000059e247 in Yosys::Pass::call (design=0x1341120, args=Warnung: Type size unknown, assuming 1. Try casting to a known type, or void *.
Warnung: Type size unknown, assuming 1. Try casting to a known type, or void *.
Python Exception <class 'gdb.error'> Cannot perform pointer math on incomplete type "std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >", try casting to a known type, or void *.: 
std::vector of length 128, capacity 128) at kernel/register.cc:316
#14 0x000000000059de7a in Yosys::Pass::call (design=0x1341120, command=...) at kernel/register.cc:293
#15 0x000000000059ebd6 in Yosys::ScriptPass::run_nocheck (this=<optimized out>, command=..., info=...) at kernel/register.cc:415
#16 0x0000000000a67300 in (anonymous namespace)::Abc9Pass::script (this=<optimized out>) at passes/techmap/abc9.cc:410
#17 0x0000000000a64713 in (anonymous namespace)::Abc9Pass::execute (this=0x12c7d98 <(anonymous namespace)::Abc9Pass>, args=Warnung: Type size unknown, assuming 1. Try casting to a known type, or void *.
Warnung: Type size unknown, assuming 1. Try casting to a known type, or void *.
Python Exception <class 'gdb.error'> Cannot perform pointer math on incomplete type "std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >", try casting to a known type, or void *.: 
std::vector of length 160, capacity 160, design=<optimized out>)
    at passes/techmap/abc9.cc:270
#18 0x000000000059e247 in Yosys::Pass::call (design=0x1341120, args=Warnung: Type size unknown, assuming 1. Try casting to a known type, or void *.
Warnung: Type size unknown, assuming 1. Try casting to a known type, or void *.
Python Exception <class 'gdb.error'> Cannot perform pointer math on incomplete type "std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >", try casting to a known type, or void *.: 
std::vector of length 160, capacity 160) at kernel/register.cc:316
#19 0x000000000059de7a in Yosys::Pass::call (design=0x1341120, command=...) at kernel/register.cc:293
#20 0x000000000059eb1a in Yosys::ScriptPass::run (this=0x12d78e0 <(anonymous namespace)::SynthIntelALMPass>, command=..., info=...) at kernel/register.cc:402
#21 0x0000000000fa0dab in (anonymous namespace)::SynthIntelALMPass::script (this=0x12d78e0 <(anonymous namespace)::SynthIntelALMPass>) at techlibs/intel_alm/synth_intel_alm.cc:289
#22 0x0000000000f9f165 in (anonymous namespace)::SynthIntelALMPass::execute (this=0x12d78e0 <(anonymous namespace)::SynthIntelALMPass>, args=..., design=0x1341120) at techlibs/intel_alm/synth_intel_alm.cc:188
#23 0x000000000059e247 in Yosys::Pass::call (design=0x1341120, args=Warnung: Type size unknown, assuming 1. Try casting to a known type, or void *.
Warnung: Type size unknown, assuming 1. Try casting to a known type, or void *.
Python Exception <class 'gdb.error'> Cannot perform pointer math on incomplete type "std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >", try casting to a known type, or void *.: 
std::vector of length 32, capacity 32) at kernel/register.cc:316
#24 0x000000000059de7a in Yosys::Pass::call (design=0x1341120, command=...) at kernel/register.cc:293
#25 0x000000000066fa6c in Yosys::shell (design=0x1341120) at kernel/yosys.cc:1274
#26 0x000000000058e9ca in main (argc=1, argv=<optimized out>) at kernel/driver.cc:553

```

[bram_wb8.zip](https://github.com/YosysHQ/yosys/files/8341087/bram_wb8.zip)

gatecat, 2022-03-24:
So, had a moment to look at this, the problem is that the M10K module has port widths depending on a parameter. It also has timing, so the xaiger backend attempts to add those, but based on the underived (ie not taking parameter into account) version. For modules with any of the abc9 box related attributes set; `abc9_ops` will create derived versions based on the parameters; but for modules with timing only this doesn't happen.

gatecat, 2022-03-24:
Can you give #3246 a try? It at least appears to build with that, I haven't actually checked the output works :D

maikmerten, 2022-03-24:
#3246 fixes the crash problem for the provided testcase and another (more complex) design with inferred M10Ks. I haven't assembled yet a proper SoC for my Cyclone-V FPGA-board so I cannot verify if the output is behaving correctly - but I'd think that faulty behavior would warrant its own issue.

Thanks you very much!

Ravenslofty, 2022-03-24:
(faulty behaviour at this level would more likely be a `nextpnr-mistral` bug anyway)

PR #3246 (2022-03-28): abc9_ops: Also derive blackboxes with timing info
Fixes #3244; as we now derive any box with timing as it could have port widths, generate blocks, etc that depend on parameters.
closed 2022-03-28 by Ravenslofty