YosysHQ/yosys #3261: synth_ice40 not working with nextpnr since Yosys 0.12

NMerkley, 2022-03-31:
## Steps to reproduce the issue
```
module test
    (
      input [2:0] NUM
    , output wire  OUTPUT
    );

  assign OUTPUT = NUM == 3'd3;
endmodule
```

`yosys test.v -o test.json -p synth_ice40`
`nextpnr-ice40 --hx1k --package tq144 --json test.json --pcf-allow-unconstrained`

## Expected behavior

nextpnr-ice40 would finish with no errors. This is what happens with yosys 0.12

## Actual behavior

nextpnr-ice40 fails with an error
`ERROR: cell type '$eq' is unsupported`

It appears that yosys is generating different cell types post-0.12 ($eq vs SB_LUT4 here) besides just $eq, but I don't know enough about yosys to know if that is expected or not

whitequark, 2022-03-31:
@clairexen There's been a constant stream of issues related to the recent `read` change. I think that, at least, `synth_*` scripts without `-top`/`-auto-top` should be deprecated; this still breaks a lot of people's workflows, but at least it no longer *silently* breaks them.

NMerkley, 2022-03-31:
While not directed at me, `-p 'hierarchy -auto-top; synth_ice40'` does seem to fix my problems, so thanks for making me aware of that

commit 8a1d531b (2022-04-07): Merge pull request #3269 from YosysHQ/micko/fix_autotop

Reorder steps in -auto-top to fix synth command, fixes #3261
commit 977002b1 (2022-04-07): Reorder steps in -auto-top to fix synth command, fixes #3261
closed 2022-04-07 by whitequark