YosysHQ/yosys #3266: Blinky from apicula does not get optimized completely

catkira, 2022-04-04:
## Steps to reproduce the issue

load this file in yosys with
read_verilog -DLEDS_NR=3 blinky.v

https://github.com/YosysHQ/apicula/blob/master/examples/blinky.v

## Expected behavior
I would expect that the 26 bit wide ctr_d wire and dff get reduced to 3 bit wide. 

## Actual behavior
The wire and register stay 26 bit wide, even after I execute opt_dff
[image]

mwkmwkmwk, 2022-04-04:
how? removing lower bits of the counter is obviously impossible here

closed 2022-04-04 by mwkmwkmwk

catkira, 2022-04-04:
Ah yes, I did not read the code correctly ^^

tcal-x, 2022-04-04:
I'm confused by the `<3>` at the _input_ to the `ctr_q` rhombus.    Is that a known quirk?

catkira, 2022-04-04:
yes, that also confused me, if only 3 bits come out of the FF, some input bits should be unused. But they are used, because the counter counts up from 0 to max_value.