YosysHQ/yosys #3278: Synthesis issue with shift

1353369570, 2022-04-12:
## Steps to reproduce the issue

Consider the following piece of code:

module top  (y, clk, wire3, wire2);
  output wire [(32'h50):(32'h0)] y;
  input wire  clk;
  input wire signed [(5'h15):(1'h0)] wire3;
  input wire signed [(2'h3):(1'h0)] wire2;
  reg signed [(5'h15):(1'h0)] reg10 = (1'h0);
  assign y = {reg10};
  always
    @(posedge clk) reg10 <= (-$signed((-(wire2 << {wire3}))));
endmodule

Run with yosys version

   Generated by Yosys 0.9+4081,Yosys 0.15 (git sha1 UNKNOWN, clang 10.0.0-4ubuntu1 -fPIC -Os)

## Expected behavior
When passing the input wire2=4'h3,wire3=4'h3; I would expect the output to give 4'h8.

## Actual behavior
The synthesised output is the following, which when givenwire2=4'h3,wire3=4'h3, gives 24'h3ffff8 as output.

module top(y, clk, wire3, wire2);
  wire _00_;
  wire _01_;
  wire _02_;
  wire _03_;
  wire _04_;
  wire _05_;
  wire _06_;
  wire _07_;
  wire _08_;
  wire _09_;
  wire _10_;
  wire _11_;
  wire _12_;
  wire _13_;
  wire _14_;
  wire _15_;
  wire _16_;
  wire _17_;
  wire _18_;
  wire _19_;
  wire _20_;
  wire _21_;
  wire _22_;
  wire _23_;
  wire _24_;
  wire _25_;
  wire _26_;
  wire _27_;
  wire _28_;
  wire _29_;
  wire _30_;
  (* force_downto = 32'd1 *)
  (* src = "reduce_identity_yosys.v:9.42-9.58|/usr/local/bin/../share/yosys/techmap.v:93.23-93.24" *)
  wire [3:0] _31_;
  (* src = "reduce_identity_yosys.v:3.30-3.33" *)
  input clk;
  wire clk;
  (* init = 22'bx0xxxxxxxxxxxxxxxxx000 *)
  (* src = "reduce_identity_yosys.v:6.31-6.36" *)
  wire [21:0] reg10;
  (* src = "reduce_identity_yosys.v:5.37-5.42" *)
  input [3:0] wire2;
  wire [3:0] wire2;
  (* src = "reduce_identity_yosys.v:4.38-4.43" *)
  input [21:0] wire3;
  wire [21:0] wire3;
  (* src = "reduce_identity_yosys.v:2.34-2.35" *)
  output [80:0] y;
  wire [80:0] y;
  assign _00_ = ~wire3[2];
  assign _01_ = wire3[4] | wire3[3];
  assign _02_ = wire3[6] | wire3[5];
  assign _03_ = _02_ | _01_;
  assign _04_ = wire3[8] | wire3[7];
  assign _05_ = wire3[10] | wire3[9];
  assign _06_ = _05_ | _04_;
  assign _07_ = _06_ | _03_;
  assign _08_ = wire3[12] | wire3[11];
  assign _09_ = wire3[14] | wire3[13];
  assign _10_ = _09_ | _08_;
  assign _11_ = wire3[16] | wire3[15];
  assign _12_ = wire3[18] | wire3[17];
  assign _13_ = _12_ | _11_;
  assign _14_ = _13_ | _10_;
  assign _15_ = _14_ | _07_;
  assign _16_ = wire3[20] | wire3[19];
  assign _17_ = _16_ | wire3[21];
  assign _18_ = _17_ | _15_;
  assign _19_ = _18_ | ~(wire2[0]);
  assign _20_ = _19_ | wire3[0];
  assign _21_ = _20_ | wire3[1];
  assign _31_[0] = _00_ & ~(_21_);
  assign _22_ = _18_ | ~(wire2[1]);
  assign _23_ = wire3[0] ? _19_ : _22_;
  assign _24_ = _23_ | wire3[1];
  assign _31_[1] = _00_ & ~(_24_);
  assign _25_ = _18_ | ~(wire2[2]);
  assign _26_ = wire3[0] ? _22_ : _25_;
  assign _27_ = wire3[1] ? _20_ : _26_;
  assign _31_[2] = _00_ & ~(_27_);
  assign _28_ = _18_ | ~(wire2[3]);
  assign _29_ = wire3[0] ? _25_ : _28_;
  assign _30_ = wire3[1] ? _23_ : _29_;
  assign _31_[3] = _00_ & ~(_30_);
  reg \reg10_reg[0]  = 1'h0;
  (* src = "reduce_identity_yosys.v:8.3-9.63" *)
  always @(posedge clk)
    \reg10_reg[0]  <= _31_[0];
  assign reg10[0] = \reg10_reg[0] ;
  reg \reg10_reg[1]  = 1'h0;
  (* src = "reduce_identity_yosys.v:8.3-9.63" *)
  always @(posedge clk)
    \reg10_reg[1]  <= _31_[1];
  assign reg10[1] = \reg10_reg[1] ;
  reg \reg10_reg[2]  = 1'h0;
  (* src = "reduce_identity_yosys.v:8.3-9.63" *)
  always @(posedge clk)
    \reg10_reg[2]  <= _31_[2];
  assign reg10[2] = \reg10_reg[2] ;
  reg \reg10_reg[20]  = 1'h0;
  (* src = "reduce_identity_yosys.v:8.3-9.63" *)
  always @(posedge clk)
    \reg10_reg[20]  <= _31_[3];
  assign reg10[20] = \reg10_reg[20] ;
  assign { reg10[21], reg10[19:3] } = { reg10[20], reg10[20], reg10[20], reg10[20], reg10[20], reg10[20], reg10[20], reg10[20], reg10[20], reg10[20], reg10[20], reg10[20], reg10[20], reg10[20], reg10[20], reg10[20], reg10[20], reg10[20] };
  assign y = { 59'h000000000000000, reg10[20], reg10[20], reg10[20], reg10[20], reg10[20], reg10[20], reg10[20], reg10[20], reg10[20], reg10[20], reg10[20], reg10[20], reg10[20], reg10[20], reg10[20], reg10[20], reg10[20], reg10[20], reg10[20], reg10[2:0] };
endmodule

I have  a  script that compares the RTL to the synthesised Verilog. To run everything, run ./run.sh.

[test_bug.zip](https://github.com/YosysHQ/yosys/files/8472538/test_bug.zip)

Ravenslofty, 2022-04-15:
If I simplify the Yosys script to `hierarchy -top top_1; proc; equiv_opt script script.ys`, with `script.ys` being `wreduce; alumacc`, then even Yosys' internal equivalence checking fails.

I hope the others won't mind me nominating this for discussion.

zachjs, 2022-04-15:
I've done a tiny bit of cleanup/reduction for this test case based on the information above.

```SystemVerilog
module top(
    output wire [4:0] out,
    input wire signed [3:0] inp1,
    input wire [3:0] inp2
);
    assign out = -$signed(-(inp1 << inp2));
endmodule
```

```
read_verilog -sv rtl.sv
proc
opt -full
eval -set inp1 3 -set inp2 3
dump
wreduce
alumacc
dump
eval -set inp1 3 -set inp2 3
```

mwkmwkmwk, 2022-04-16:
This has nothing to do with shift, even:

```
module top(
    output wire [4:0] out,
    input wire signed [3:0] inp1,
);
    assign out = -$signed(-(inp1 + 4'sh0));
endmodule
```

```yosys t2.v -f verilog -p 'opt_expr; wreduce; dump; eval -set inp1 8; alumacc; dump; eval -set inp1 8'```

There's a bug somewhere in `alumacc` having to do with merging two negations with this width configuration.

pmnmalo, 2026-10-21:
This still reproduces in Yosys 0.70 (28ba3cb92). We hit it through random differential testing (a VlogHammer-style generator), where the netlist of `-($signed(-(s4 ^~ 4'sb1110)))` disagreed with simulation.

A smaller case, with no shift or add inside, only nested negations:

```verilog
module t(input signed [4:0] a, output [15:0] y);
  assign y = -($signed(5'(-a)));
endmodule
```

With `a = -16`, the inner `-a` wraps at 5 bits back to -16, and the outer negation at 16 bits gives 16:

```
read_verilog -sv t.v; proc; opt;                     eval -set a 16 -show y   -> 16'0000000000010000  (right)
read_verilog -sv t.v; synth -top t;                  eval -set a 16 -show y   -> 16'1111111111110000
read_verilog -sv t.v; proc; opt; wreduce; alumacc;   eval -set a 16 -show y   -> 16'1111111111110000
read_verilog -sv t.v; proc; opt; alumacc;            eval -set a 16 -show y   -> 16'0000000000010000
read_verilog -sv t.v; proc; opt; wreduce;            eval -set a 16 -show y   -> 16'0000000000010000
```

After `wreduce`, the outer `$neg` reads the 5-bit result of the inner one directly (A_WIDTH 5, Y_WIDTH 16). `alumacc` then folds both into a single `$macc_v2` addend `+a` at 16 bits, dropping the inner negation's 5-bit wrap. In our tests only `$neg` is affected this way: a truncated `$add` or `$sub` in the same position keeps its width. As a workaround, our front end now writes `-x` as `0 - x`, which synthesizes correctly.

pmnmalo, 2026-10-21:
I traced this in current master (0.70+163, 3718f7f13). The cause is in `macc_may_overflow()` (passes/techmap/alumacc.cc), which `merge_macc()` uses to decide whether a narrower `$macc` can be merged into a wider one.

It sizes each term as an unsigned value: a k-bit term counts as up to 2^k − 1, and the sum is checked against the narrower width. For a signed term that is subtracted, that is one bit short. Negating the most negative value, −2^(k−1), gives +2^(k−1), which needs k+1 signed bits. So a single negated signed term of full width (`-a`, with `a` signed and 5 bits wide, inside a 5-bit result) is judged unable to overflow. After `wreduce` has narrowed the outer cell to read those 5 bits, the inner `$neg` is merged, and its wrap is lost.

With two terms the sum already crosses the width (31 + 31 needs 6 bits), so `-a - b` is not merged and comes out right. The error needs the count to land exactly on the width.

Counting k+1 bits for a signed term with `do_subtract` makes both examples here come out right, in `wreduce; alumacc` and in `synth`. `equiv_opt -assert wreduce; alumacc` then proves them, and the scripts in tests/alumacc and tests/arith_tree still pass. I'm leaving the fix itself to you, as your contribution policy asks.