YosysHQ/yosys #3283: Removing a redundant (illegal?) line makes result slower

tommythorn, 2022-04-17:
## Steps to reproduce the issue

The code is on the `buggy-is-faster` branch of  https://github.com/tommythorn/yarvi

Intel's Quartus error on this because I have a `s6_branch_taken <= s5_branch_taken;`
assignment repeated in two different always blocks.  Obviously a mistake.  However
yosys+nextpnr  (Yosys 0.13+3 (git sha1 61324cf55, gcc 11.2.0-7ubuntu2 -fPIC -Os))
consistently gives a worse result for this.  Over 15 seeds it went from (as produced
with `make fmax`)

```
50.30 MHz (FAIL at 148.02 MHz)
51.19 MHz (FAIL at 148.02 MHz)
60.09 MHz (FAIL at 148.02 MHz)
60.52 MHz (FAIL at 148.02 MHz)
61.26 MHz (FAIL at 148.02 MHz)
61.70 MHz (FAIL at 148.02 MHz)
62.19 MHz (FAIL at 148.02 MHz)
62.31 MHz (FAIL at 148.02 MHz)
62.35 MHz (FAIL at 148.02 MHz)
62.81 MHz (FAIL at 148.02 MHz)
63.02 MHz (FAIL at 148.02 MHz)
63.44 MHz (FAIL at 148.02 MHz)
63.72 MHz (FAIL at 148.02 MHz)
63.78 MHz (FAIL at 148.02 MHz)
65.65 MHz (FAIL at 148.02 MHz)
```
to
```
49.43 MHz (FAIL at 148.02 MHz)
50.74 MHz (FAIL at 148.02 MHz)
57.03 MHz (FAIL at 148.02 MHz)
57.15 MHz (FAIL at 148.02 MHz)
57.69 MHz (FAIL at 148.02 MHz)
57.75 MHz (FAIL at 148.02 MHz)
58.57 MHz (FAIL at 148.02 MHz)
59.26 MHz (FAIL at 148.02 MHz)
59.49 MHz (FAIL at 148.02 MHz)
60.41 MHz (FAIL at 148.02 MHz)
60.68 MHz (FAIL at 148.02 MHz)
60.72 MHz (FAIL at 148.02 MHz)
61.21 MHz (FAIL at 148.02 MHz)
61.27 MHz (FAIL at 148.02 MHz)
61.49 MHz (FAIL at 148.02 MHz)
```
It seems like a bug to me.

whitequark, 2022-04-17:
You can do anything really fast if you don't care about doing it correctly.

tommythorn, 2022-04-17:
I'm not sure what I can do with that statement though.  Yosys doesn't complain about the code in the branch show, and both verilator and Icarus Verilog are happy and does the right thing.  I only discovered it because Intel's Quartus didn't want to have anything to do with it.  However deleting either of the redundant lines hurts the performance which is insanely frustrating as I don't see why it would any difference at all and the unpredictability of the tools makes it basically impossible to improve performance.

whitequark, 2022-04-17:
Assigning to the same register from two different `always` blocks is incorrect in Verilog (no matter which tool you're using), so the performance of it shouldn't matter to you.

tommythorn, 2022-04-17:
IOW, yosys can have it both ways.  Either the code is illegal and should have been refuse or it's fine and the results should have been the same.

whitequark, 2022-04-17:
> Yosys should refuse the code but doesn't.

I agree. Yosys' Verilog frontend is well known to accept so much illegal code that there hasn't even been an effort to enumerate it. The answer Claire gave me a few years ago when I had the same concerns is that you're expected to only feed Yosys code that is accepted by Verilator.

tommythorn, 2022-04-17:
FWIW, as I mention both Verilator and Icarus accepts this, which is why I didn't catch it until I tried Quartus.  In hindsight it's obviously a bug in my code but I was confused because only Quartus complained.  Thanks for the replies!

closed 2022-04-17 by tommythorn