YosysHQ/yosys #3286: Can I easily concatenate multiple signals?

mmxsrup, 2022-04-20:
I want to create a path to create a signal that concatenates multiple selected signals. I know that we can use `addConcat()` to create two signals into one signal. I can simply concatenate `n` signals by naively calling `addConcat()` `n-1` times.
https://github.com/YosysHQ/yosys/blob/93508d58dafbbffcedffa70b21a197b6fca8bb30/kernel/rtlil.h#L1306

Do I need to use `n-1` concat cells to create a signal that concatenates `n` signals?  If not, is there a more convenient cell or function that allows me to concatenate n signals at one time?

MidnightSkylight, 2022-04-21:
So in this case what I would actually recommend doing is to define this as a connection. You can define a signal with a width of the concatenated construct, then define that signal to be equivalent to a connection of each of the concatenated signals. The construct that you found defines a *cell* that performs concatenation which is almost certainly not what you actually want. 

See: https://github.com/YosysHQ/yosys/blob/68c67c40ec75b192f4f1be9711afe0df8973e797/frontends/ast/genrtlil.cc#L1553-L1560

mmxsrup, 2022-04-24:
It could simply be implemented using the `append` method. Thank you!
```cpp
RTLIL::SigSpec append_sigs(RTLIL::Module *module, std::vector<SigSpec> &sigs) {
	RTLIL::SigSpec sig_result;
	for (auto sig : sigs) {
		sig_result.append(sig);
	}
	return sig_result;
}
```

closed 2022-04-24 by mmxsrup