YosysHQ/yosys #3295: `synth_ecp5` with `-abc9` fails since 8b64dc1dcec8b13baf46679a21e14ace1241f894

hannesha, 2022-04-30:
Running `git bisect` with the last good version from early march, hints at 8b64dc1dcec8b13baf46679a21e14ace1241f894 being the first bad commit.
Reverting it restores the expected synthesis behavior.

## Steps to reproduce the issue

Run `make poc.json` inside the `poc` folder.

## Expected behavior

Synthesis should complete normally,

## Actual behavior

Synthesis doesn't complete and prints the following error:
```ERROR: Module `\i2s' referenced in module `\top' in cell `\i2s_out' is not part of the design.```

Running `synth_ecp5` without `-abc9` still works as expected.

### Example
`poc/yosys_broken.log` contains the debug log of the latest build.
`poc/yosys_reverted.log` contains the debug log of current master with 8b64dc1dcec8b13baf46679a21e14ace1241f894 reverted.
[poc.tar.gz](https://github.com/YosysHQ/yosys/files/8597721/poc.tar.gz)

gatecat, 2022-05-01:
Thanks for the report and for including a testcase! I hope that #3296 fixes this.

hannesha, 2022-05-01:
Yes this fixes synthesis. It even works for a larger design that had the same error.

Ravenslofty, 2022-05-23:
Fixed by #3296.

closed 2022-05-23 by Ravenslofty