YosysHQ/yosys #3308: Yosys  output and input design equivalence is inconsistent

1353369570, 2022-05-10:
## Steps to reproduce the issue
Consider the following piece of code:
 module top  ( var_ah, var_ab, var_ac ); 
     output var_ah ; 
    input var_ab, var_ac ; 
 assign var_ah  = var_ac ? var_ab : 1'bz ; 
 endmodule 

Run with yosys version

Generated by Yosys 0.9+4081,Yosys 0.15 (git sha1 UNKNOWN, clang 10.0.0-4ubuntu1 -fPIC -Os)

and the following command

 yosys -p 'read_verilog  rtl.v; synth; write_verilog  syn.v'

## Expected behavior

When passing the input var_ac=0; I would expect the output to give1'bz.

## Actual behavior

*The synthesised output is the following, which when given var_ac=0, gives var_ab as output.At the same time, symbiyosys also found that the output and input are not equivalent when checking their equivalence  .

yosys output is as following：
/* Generated by Yosys 0.15 (git sha1 UNKNOWN, gcc 9.4.0-1ubuntu1~20.04 -fPIC -Os) */

(* src = "rtl.v:2.2-7.11" *)
module top(var_ah, var_ab, var_ac);
  (* src = "rtl.v:4.11-4.17" *)
  input var_ab;
  wire var_ab;
  (* src = "rtl.v:4.19-4.25" *)
  input var_ac;
  wire var_ac;
  (* src = "rtl.v:3.13-3.19" *)
  output var_ah;
  wire var_ah;
  assign var_ah = var_ab;
endmodule

there is a script that compares the RTL to the synthesised Verilog. to run sby proof.sby.

[test_bug.zip](https://github.com/YosysHQ/yosys/files/8661186/test_bug.zip)

1353369570, 2022-05-28:
Can you help me solve this problem? I want to know the cause of this problem

nakengelhardt, 2022-05-28:
Neither Yosys nor the formal solvers can use tristate logic, so this is not expected to work.

closed 2022-05-28 by nakengelhardt