YosysHQ/yosys #3345: Feature request: Variables of data type real in initial construct

daglem, 2022-05-29:
Verilog 2005 specifies the variable data types integer, real, realtime, reg, and time, of which Yosys seems to have full support for integer, and partial support for real (parameter real).

It would be very useful if the support for real were to be extended to initial constructs, i.e.

```
initial begin
    real r = 10.0;
    ...
end
```

This currently fails with the error "ERROR: syntax error, unexpected TOK_REAL".

Allowing real in this setting would facilitate parametrization of modules which depend on iterative floating point calculations, e.g. to initialize BRAM. It also seems like a logical enhancement, since Yosys already supports real math functions, i.e. $sin(x) et al.

nuvious, 2023-02-07:
Seeing the same problem; love the tool btw!
Here's the code I was using that manifested the error. Working fine in iverilog.

```verilog
module bhaskara_sin;
    input real theta;
    output real sin;
    function real sin_func;
        input real x;
        real numerator;
        real denominator;
      
        real PI 3.14159265359

        begin
            numerator = PI;
            numerator = 16 * x * (PI - x);
            denominator = (5 * PI * PI) - (4 * x * (PI - x));
            sin_func = numerator / denominator;
        end
    endfunction

    always @ * begin
        sin = sin_func(theta);
    end
endmodule

```

nakengelhardt, 2023-02-08:
@nuvious Well that's just non-synthesizable code, you're using `real` as input and output port types, not as parameters. Very different from the original issue.