YosysHQ/yosys #3370: yosys produces more LUTs

splinedrive, 2022-06-09:
Hi,
I have written a soc for icestick the design will only fit for yosys 0.17 my current oss-build suite produces >=16 % bigger design.
You can reprodes it via my soc
git clone https://github.com/splinedrive/kianRiscV
. ./oss-cad-suite-build/set_toolchain.sh
cd kianRiscV/kianv_harris_mcycle_edition/gateware
./build_stick.sh 

soc will fit on icestick with yosys 0.17

Please help!

nakengelhardt, 2022-06-09:
No, it would not be nice to make the design behave wrong by default just because it uses less resources. I once lost a whole year trying to track down such an issue, so I feel very strongly about that!

splinedrive, 2022-06-09:
Could you give me a verilog example?

closed 2022-06-09 by splinedrive

nakengelhardt, 2022-06-10:
I'll try and write a blog post about it since a few people have asked about this, but I'm swamped at the moment so it might take a bit...

splinedrive, 2022-06-10:
@nakengelhardt  nice!

fontamsoc, 2022-06-16:
@nakengelhardt beside `-no-rw-check`, are there other options that add extra circuitry for safety ?

In fact, using `-no-rw-check` in one of my project resulted in a smaller design and still working design; I am wondering whether there are other options I could use to reduce logic usage:

[image]

nakengelhardt, 2022-06-24:
We usually try very hard not to make unsound optimizations, even optionally. If you do find one, please report it, as that would be a bug. We only grandfathered it in here because other tools also expect the user to have special consideration for collision behavior, and thus there is already a large amount of existing code where the collision behavior was verified independently. For new code, please write the memory pattern that corresponds to the actually desired implementation.

But for the malicious compliance answer: have you considered using `bugpoint` to delete random pieces of your design as long as they don't make your testbench fail?