YosysHQ/yosys #3376: Synthesis Error when the FIR filter Verilog file is altered

Jingfan2007, 2022-06-14:
## Steps to reproduce the issue

We came across this problem when trying to reproduce the results in ABACUS: A Technique for Automated Behavioral Synthesis of Approximate Computing Circuits. One of the files generated by ABACUS was to directly assign one of the product taps to zero, and the area was calculated by using the Yosys script provided by ABACUS namely "yosys.script". The command used is 

> yosys -s yosys.script

There are two Verilog files, one is the fir_filter_original.v, and the other is fir_filter_altered.v, the only difference is as shown in the figure below,

[image]
[image]

Please change the wanted file name to  fir_filter.v to measure the data of those circuits.

[synth_problem.zip](https://github.com/YosysHQ/yosys/files/8902044/synth_problem.zip)

## Expected behavior

We used DC to check the area of those two and since then it is the approximation of the circuit, the area was reduced from 8026 to 7850. We believe that this is more reasonable.

## Actual behavior

According to Yosys, the original circuit has an area of 66005.1678, and the altered one 164.7243. The area was largely reduced and this is unreasonable. Therefore we produce the .blif file of the circuit and checked that in the altered file, the input wasn't used inside the circuit.

jix, 2022-06-14:
The file "fir_filter_altered.v" does not parse and Yosys outputs "./fir_filter.v:37: ERROR: syntax error, unexpected end of file". This is due to the final character on the line not being a semicolon but the similar looking Unicode character U+FF1B (FULLWIDTH SEMICOLON).

Changing this to a semicolon and running it with the current git master of Yosys does not reproduce this issue, the areas reported by the script change from "48071.806900" to "46512.792300".

mwkmwkmwk, 2022-06-14:
You are using yosys version 0.7, which is 6 years old. The current yosys version doesn't exhibit this problem. Please upgrade.

closed 2022-06-14 by mwkmwkmwk